openbmc/openbmc #10: Unable to boot using GCC 5.2.0

bradbishop, 2015-10-01:
At the moment, we are using Yocto master head which generates a GCC 5.2.0 based toolchain.  This toolchain doesn't produce a usable kernel on Palmetto:

AST2400EVB>bootm 0x50000000
## Booting image at 50000000 ...
   Image Name:
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2312568 Bytes = 2.2 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
OK

Starting kernel ...

atags_to_fdt
atags=40000100
fdt=4023bb48
atags_to_fdt
atags=4023bb48
fdt=40772bc8
Uncompressing Linux... done, booting the kernel.

[ hang here]

If I back off to GCC 4.9.3 it works fine.

williamspatrick, 2016-04-01:
Has this been resolved?

mdmillerii, 2016-04-01:
The release notes delta in the yocto-2.0.1 update included refs to these bugs:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65932
(fixed in 6 backported to gcc 5.4)

And  kernel workaround (part of linux 4.2)
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit?id=a077224fd35b2f7fbc93f14cf67074fc792fbac2

shenki, 2016-04-04:
More bugs:

https://github.com/openbmc/ipmi-fru-parser/issues/12

https://github.com/openbmc/u-boot/issues/5

williamspatrick, 2016-07-27:
Fixed.

closed 2016-07-27 by williamspatrick