openbmc/openbmc #20: reboot -f hang

bradbishop, 2015-10-19:
reboot -f is hanging:

root@palmetto:~# reboot -f
Rebooting.reboot: Restarting system

I tried the 'reboot=force' cmdline option but that didn't seem to help any.  I'd strace it but strace isn't in the build.  Does it make sense to add that?

jk-ozlabs, 2015-10-19:
`strace` won't help with this, but it might be useful in general.

shenki, 2015-10-19:
@bradbishop Were you on the serial console, or ssh'd in?

bradbishop, 2015-10-19:
serial console

nkskjames, 2015-10-22:
reboot without the -f hangs too for what it's worth.

shenki, 2015-10-22:
@jk-ozlabs did some investigation and found that setting WDT_CTRL_RESET_MODE_FULL_CHIP in the watchdog resolved the issue.

https://github.com/jk-ozlabs/linux/commit/2f1711908ac92968e2d533616c4cdb8d7b848677

I'm unsure why we now need this flag, when reboot was working before.

jk-ozlabs, 2015-10-28:
My money is on magic SCU register configuration.

shenki, 2015-11-24:
On ```Linux barreleye 4.3.0-openbmc-20151123-1 #1 Mon Nov 23 16:15:26 UTC 2015 armv5tejl GNU/Linux``` with OpenBMC userspace, plus the userspace tool `watchdog` running:

```
echo 8 > /proc/sys/kernel/printk
echo "file aspeed_wdt.c +p" > /sys/kernel/debug/dynamic_debug/control
```

```
[  OK  ] Stopped target Network.
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped target Multi-User System.
         Stopping Phosphor OpenBMC BT to DBUS...
         Stopping Kernel Logging Service...
         Stopping Phosphor OpenBMC IPMI d         Stopping Serial Getty on ttyS4...
         Stopping DBUS introspecting REST server....
         Stopping SYSV: A software watchdog...
         Stopping Temp placeholder for skeleton function...
         Stopping Network Service...
         Stopping System Logging Service...
         Stopping Phosphor OpenBMC DBus REST daemon...
         Stopping Phosphor OpenBMC event management daemon...
         Stopping Login Service...
[  OK  ] Stopped target Timers.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[  OK  ] Stopped target Remote File Systems.
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
[  OK  ] Stopped Phosphor OpenBMC IPMI daemon.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped DBUS introspecting REST server..
[  OK  ] Stopped Temp placeholder for skeleton function.
watchdog watchdog0: ping
[  OK  ] Stopped Phosphor OpenBMC event management daemon.
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Phosphor OpenBMC BT to DBUS.
[  OK  ] Stopped Serial Getty on ttyS4.
[  OK  ] Stopped SYSV: A software watchdog.
[  OK  ] Stopped Phosphor OpenBMC DBus REST daemon.
         Stopping Phosphor OpenBMC DBus service management daemon...
[  OK  ] Removed slice system-serial\x2dgetty.slice.
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped Phosphor OpenBMC DBus service management daemon.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed Syslog Socket.
[  OK  ] Closed dropbear.socket.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped Dispatch Password Requests to Console Directory Watch.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Stopped target System Initialization.
[  OK  ] Stopped target Swap.
         Stopping Load/Save Random Seed...
         Stopping Network Time Synchronization...
         Stopping Update UTMP about System Boot/Shutdown...
[  OK  ] Stopped Update is Completed.
[  OK  ] Stopped Rebuild Journal Catalog.
[  OK  ] Stopped Rebuild Hardware Database.
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /var/volatile...
         Unmounting Temporary Directory...
[  OK  ] Unmounted /var/volatile.
[  OK  ] Unmounted Temporary Directory.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Create System Users.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
watchdog watchdog0: starting with timeout of 5 (rate 48000000)
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[420]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
watchdog watchdog0: starting with timeout of 5 (rate 48000000)
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
watchdog watchdog0: ping
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Rebooting.
reboot: Restarting system
```

And then the system hangs.

It appears that systemd is bringing down the watchdog daemon here `[  OK  ] Stopped SYSV: A software watchdog.`, and from then on we don't get any pings.

The watchdog is then configured, twice and then we get a ping. What is causing this ping?

```
[  OK  ] Reached target Shutdown.
watchdog watchdog0: starting with timeout of 5 (rate 48000000)
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[420]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
watchdog watchdog0: starting with timeout of 5 (rate 48000000)
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
watchdog watchdog0: ping
systemd-shutdown[1]: Unmounting file systems.
```

shenki, 2015-11-24:
```
root@barreleye:~# watchdog -F -v
watchdog: startiwatchdog watchdog0: starting with timeout of 5 (rate 48000000)
ng daemon (5.14):
watchdog: int=1s realtime=yes sync=no soft=nowatchdog watchdog0: ping
 mla=0 mem=0
wawatchdog watchdog0: ping
tchdog: ping: nowatchdog watchdog0: ping
 machine to checwatchdog watchdog0: ping
k
watchdog: filwatchdog watchdog0: ping
e: no file to chwatchdog watchdog0: ping
eck
watchdog: pwatchdog watchdog0: ping
idfile: no server process to check
watchdog: interface: no interface to check
watchdog: temperature: no sensors to check
watchdog: test=none(0) repair=none(0) alive=/dev/watchdog heartbeat=none to=root no_act=no force=no
watchdog: cannot set timeout 60 (errno = 22 = 'Invalid argument')
watchdog: hardware watchdog identity: aspeed_wdt
watchdog: still watchdog watchdog0: ping
alive after 1 inwatchdog watchdog0: ping
terval(s)
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog: still watchdog watchdog0: ping
alive after 2 inwatchdog watchdog0: ping
terval(s)
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog: still watchdog watchdog0: ping
alive after 3 inwatchdog watchdog0: ping
terval(s)
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog: still watchdog watchdog0: ping
alive after 4 inwatchdog watchdog0: ping
terval(s)
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog: still watchdog watchdog0: ping
alive after 5 inwatchdog watchdog0: ping
terval(s)
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
watchdog watchdog0: ping
^Cwatchdog watchdog0: watchdog did not stop!
watchdog watchdog0: ping
```

And then the system was hung.

shenki, 2015-12-02:
In my testing I have added print statements to the watchdog code. It appears that systemd activates the watchdog as it shuts down userspace, as a precaution against a hang at shutdown time. It was suspected that this interaction was causing issues, but that is not the case.

I used some userspace code to open /dev/watchdog0, and after patting it a few times, let it expire. This causes the same hang behaviour as shutting down does.

I booted a the kernel with a simplified userspace that didn't perform any actions. Reboot worked fine in this case.

`/dev/mem` is currently exposed as a temporary solution for userspace to poke at some memory mapped peripherals. There are various bits of userspace that poke at peripheral registers using this interface. In order to eliminate them as a source of the bug, I booted an unmodified OpenBMC userspace with the `/dev/mem` character device disabled in the kernel config.  Reboots now work.

The next step is to work out which setting is the one that is being changed that causes our bug.

nkskjames, 2015-12-02:
So its 1 bit causing the issue.   bit 4 in SCU reg 0x70 which sets the SPI addressing mode.    I had started with devmems from bringup which use a 16MB SPI flash.  We later switched to 32MB flash which needs 4 byte addressing mode.  This bit was overriding to 3 byte addressing mode which since the BMC is already booted when this gets set, it didn't affect anything, until a reboot....

nkskjames, 2015-12-02:
I'm not sure why we are setting that register to begin with.  it is setup properly by strapping.  fix is to remove entire register set.

closed 2015-12-03 by nkskjames