openbmc/openbmc #28: stdout not captured in journald log for subprocesses of system_manager

nkskjames, 2015-10-21:
system_manager launches required dbus processes and short lived processes.   The stdout from those processes is not captured in journald log.   Need an alternative method.

williamspatrick, 2015-10-22:
Thinking about this a little bit I'm surprised it doesn't already work.  The system_manager is launched from systemd so its stdout and stderr route to journald.  When it creates a sub-process, don't the children inherit the same stdout / stderr?  Can you tell what the stdout / stderr point to for the children?  Use /proc/<pid>/fd entries to identify.

nkskjames, 2015-10-22:
I found issue.   there is a known issue with systemd not playing nicely with python stdout buffering.   so adding -u to python command line says to not buffer and now stdout shows up in journald log.

closed 2015-10-22 by nkskjames