openbmc/openbmc #29: can't include i2c-tools without perl

nkskjames, 2015-10-21:
The i2c-tools recipe has a RDEPENDs on perl because of some utility perl scripts.  we don't need those scripts, so I removed scripts and dependency on Perl, but bitbake still thinks there is a dependency.  A "smart install" maybe sees /usr/bin/perl in scripts and assumes it needs it?

williamspatrick, 2015-10-22:
Can you explain what you already tried so that we don't need to rediscover the approach that doesn't work?

Did you remove the dependency on the i2c-tools-misc package from i2c-tools?  I would be surprised if bitbake is identifying perl scripts to install.

nkskjames, 2015-10-23:
i removed the whole misc package:
#PACKAGES =+ "${PN}-misc"
#FILES_${PN}-misc = "${sbindir}/i2c-stub-from-dump \
#                        ${bindir}/ddcmon \
#                        ${bindir}/decode-edid \
#                        ${bindir}/decode-dimms \
#                        ${bindir}/decode-vaio \
#                       "
#RDEPENDS_${PN} += "${PN}-misc"
#RDEPENDS_${PN}-misc += "perl"

Then:
rm -rf build/tmp
bitbake obmc-phosphor-image
and I get this error:

ERROR: Unable to install packages. Command '/home/njames/openbmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin/smart --log-level=warning --data-dir=/home/njames/openbmc/openbmc/build/tmp/work/palmetto-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/var/lib/smart install -y packagegroup-core-ssh-dropbear@all packagegroup-base-extended@palmetto run-postinsts@all packagegroup-obmc-phosphor-apps-chassis-mgmt@all packagegroup-obmc-phosphor-apps-event-mgmt@all packagegroup-obmc-phosphor-apps-policy-mgmt@all packagegroup-obmc-phosphor-apps-fan-mgmt@all packagegroup-obmc-phosphor-apps-flash-mgmt@all bash@armv5e packagegroup-obmc-phosphor-apps-extras@all packagegroup-obmc-phosphor-apps-sensor-mgmt@all shadow@armv5e packagegroup-obmc-phosphor-apps-system-mgmt@all base-passwd@armv5e packagegroup-core-boot@palmetto locale-base-en-us@armv5e' returned 1:
Loading cache...
Updating cache...               ######################################## [100%]

Computing transaction...error: Can't install i2c-tools-3.1.2-r0@armv5e: no package provides /usr/bin/perl

My suspicion is that smart install application is inspecting the files and seeing the #!/usr/bin/perl at top of file.  Only thing I know to try is to patch out the first line in the perl scripts.

nkskjames, 2015-10-24:
Changing to this works:
PACKAGES =+ "${PN}-misc"
FILES_${PN}-misc = "${sbindir}/i2c-stub-from-dump \
                        ${bindir}/ddcmon \
                        ${bindir}/decode-edid \
                        ${bindir}/decode-dimms \
                        ${bindir}/decode-vaio \
                       "
#RDEPENDS_${PN} += "${PN}-misc"
RDEPENDS_${PN}-misc += "perl"

closed 2015-10-24 by nkskjames