openbmc/openbmc #35: Can not build image.

Kenthliu, 2015-10-23:
Hi IBM team,
I'm Foxconn BMC FW engineer Ken.
It's my pleasure to co-work with you.
We download zip in the right side, and get an openbmc-master folder.
Under openbmc-master folder, we follow the instruction executed .openbmc-env.
But just see the below error message.

openpower@openpower-VirtualBox:~/openbmc-master$ ./openbmc-env 
### Initializing OE build env ###
yocto-poky/oe-init-build-env: 47: .: Can't open /home/openpower/openbmc-master/scripts/oe-buildenv-internal

Is there anything we miss?
Do we need to use git clone?
Thanks.

williamspatrick, 2015-10-23:
Two issues:

1. You can build directly from the extracted directory, but the recommendation from bitbake is to go to a new empty directory.  When you run the command to set up the bitbake environment it will create a 'build' subdirectory in your current location.  If you do this in an empty directory it doesn't pollute your code repository / zip extraction at all.

2. It should be <dot><space> not <dot><slash>: ". openbmc-env" and not "./openbmc-env".  This is a shell operation that means "source the file" and not "run the file".

Kenthliu, 2015-10-26:
Hi William,
We can build succefully.
Please close this issue.
Thank you very much!

closed 2015-10-26 by Kenthliu