openbmc/openbmc #41: Add kernel config to board layers

shenki, 2015-10-27:
We should be keeping the Linux kernel configuration in the board specific layers.

We will leave the `aspeed_defconfig` for Kernel developers to customise to their needs, and ensure that the systems have a minimal, secure, configuration.

I'm unsure on the poky bits that need to be done here. If @bradbishop could move the `aspeed_defconfig` config into the tree, I will then follow up with any configuration changes I think we need.

bradbishop, 2015-10-30:
Done.  https://github.com/openbmc/openbmc/pull/50

bradbishop, 2015-12-15:
@shenki can this get closed?

shenki, 2015-12-15:
Have we checked that the config only enables the bits that we need? I can take a look if you want.

I think it would make sense to put the configs in the one file instead of having the hwmon chunk separate.

williamspatrick, 2016-04-01:
Any idea how this is different from #56 ?

bradbishop, 2016-04-14:
The concept is the same but the mechanics of doing it in recipes is different.

williamspatrick, 2016-06-15:
From a design stand-point, Brad and I would prefer config snippets that are based on function being enabled and not always strictly board or SoC specific.  We have ran into an issue recently where the systemd config options were not correctly enabled and it would make more sense to automatically include those config options whenever we enable systemd.

We understand that there is a desire by the kernel developers to be able to quickly see what the config is being used.  A compromise there might be to generate the config and publish it along with our reference flash images.

williamspatrick, 2016-06-30:
@bradbishop suggested we should close this because it is done and create a new story for a minimal defconfig.  @shenki - any issues with that?

shenki, 2016-07-01:
Ok. Please assign me to the new issue.

closed 2016-07-01 by shenki