openbmc/openbmc #46: sdk failes to compile against palmetto

causten, 2015-10-29:
without exporting the templateconf it works fine but fails to build if pointing towards a palmetto conf

export TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-palmetto/conf  
. oe-init-build-env
bitbake -c populate_sdk obmc-phosphor-image
results in...

NOTE: Executing RunQueue Tasks
ERROR: Function failed: do_install (log file is located at /home/causten/openbmc-barreleye/build/tmp/work/palmetto-openbmc-linux-gnueabi/linux-obmc/4.2+gitAUTOINC+0e0ca76818-r0/temp/log.do_install.10261)
ERROR: Logfile of failure stored in: /home/causten/openbmc-barreleye/build/tmp/work/palmetto-openbmc-linux-gnueabi/linux-obmc/4.2+gitAUTOINC+0e0ca76818-r0/temp/log.do_install.10261
Log data follows:
| DEBUG: Executing python function package_get_auto_pr
| DEBUG: Python function package_get_auto_pr finished
| DEBUG: Executing shell function do_install
| NOTE: make -j 4 DEPMOD=echo INSTALL_MOD_PATH=/home/causten/openbmc-barreleye/build/tmp/work/palmetto-openbmc-linux-gnueabi/linux-obmc/4.2+gitAUTOINC+0e0ca76818-r0/image modules_install
|   DEPMOD  4.2.0
| install: cannot stat 'arch/arm/boot/cuImage': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_install (log file is located at /home/causten/openbmc-barreleye/build/tmp/work/palmetto-openbmc-

Kenthliu, 2015-10-30:
The root cause may related to SRCREV in meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.2.bb

bradbishop, 2015-11-02:
I'm pretty sure this is fixed by https://github.com/openbmc/openbmc/commit/4f1decd6c7a4487e813c711b49bf675a62c489d4

closed 2015-11-07 by causten