openbmc/openbmc #92: Every core and dimm not populated shows fault

causten, 2015-11-20:
"/org/openbmc/inventory/system/chassis/motherboard/cpu0/core8": {
      "fault": "True", 
      "fru_type": "CORE", 
      "is_fru": 0, 
      "present": "False", 
      "version": "None"
    },

The core is not present nor will it ever.  Why set the Fault value to True?

williamspatrick, 2015-11-20:
The open-power/hostboot code sends it down that way from IPMI.

dcrowell77, 2016-01-11:
Can you be more specific about where you are detecting the 'fault' setting?  What bit(s) are you looking at?  Which sensor type are you looking at (we have both functional and fault sensors)?  For a non-present part, we deassert the present offset and assert the disabled offset.  Is the disabled offset the problem?

causten, 2016-02-19:
Looks promising, just need to get this version on to a system

nkskjames, 2016-03-03:
This is fixed.

closed 2016-03-03 by nkskjames