openbmc/openbmc #93: Fan Presence Detect

causten, 2015-11-20:
BMC based devices not controlled by the host do not get presence.  True for fans, system and chassis too

curl -k https://<ip>/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "None", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "Inactive", 
      "version": "None"
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "None", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "Inactive", 
      "version": "None"
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "None", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "Inactive", 
      "version": "None"
    }

causten, 2016-05-27:
Tag 0.8.1 now has presence set to True for Chassis, System and IO Board.  That is good news.  The Fan inventory items still do not show presence status.

anoo1, 2016-05-27:
@Kenthliu will take a look, his team may have a code change for this.

Kenthliu, 2016-05-31:
Do you mean add 'present' : "True" in Barreleye.py for fan?

causten, 2016-05-31:
There needs to be code added that determines if the fan is present.  I don't think it will be in the Barreleye.py file.  Likely in some fan init code

williamspatrick, 2016-06-15:
There may be a short-term fix needed for Barreleye.  This issue we are moving to 2.0 for long-term handling of BMC-owned inventory items.

sivassrr, 2016-06-29:
With level 125, Still sees failure in Palmetto as well for DIMMS. 
rg.openbmc.InventoryItem

    setPresent(string present) →
    setFault(string fault) →
    update(dict {string → variant} data) →
    fault = False
    fru_type = DIMM
    is_fru = 1
======================
For all the DIMMs and CORES as well.

williamspatrick, 2016-06-29:
@sivassrr - Problems with DIMMs and Cores is a different issue.  Please open something separate to track that.  This problem @causten is a global problem with regards to state of BMC-owned devices.  We do not presently have any code to set the present state of BMC-owned devices.

williamspatrick, 2016-06-30:
Assign to @adamliyi .

adamliyi, 2016-07-05:
Tested on Palmetto using OBMC-V1.0, system, chassis, fans FRU's present bit is still "False":

```
curl -c cjar -b cjar -k https://192.168.0.60/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan4": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    },"
```

williamspatrick, 2016-07-05:
Correct.  I expect a code change required on this one.

adamliyi, 2016-07-06:
Palmetto.py does not invoke phosphor-read-eeprom to read VPD into Inventory. So we need to test on Barreleye. On Barreleye, System, Chassis "present" property is "TRUE". Fans "present" property is "False". This issue is to fix Fans "present" property on Barreleye.
```
curl -c cjar -b cjar -k https://9.3.23.30/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/bios": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan4": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan5": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
```

adamliyi, 2016-07-06:
To fix the Fan "present" issue:
@Kenthliu 's PR: https://github.com/openbmc/skeleton/pull/59 
and 
johnhcwang's patch: https://github.com/johnhcwang/skeleton/commit/02df39a32878ed58f5919ebe7b1073ec15bedffe
should be able to fix this issue.

williamspatrick, 2016-07-06:
Palmetto doesn't have a VPD chip on the backplane that the BMC can access. That is why it is missing the call you mentioned.

williamspatrick, 2016-07-06:
openbmc/skeleton#59 needs quite a bit of clean up before it can be merged.

adamliyi, 2016-07-07:
@Kenthliu said he will clean up and re-submit openbmc/skeleton#59, so this issue is currently depend on this PR.

williamspatrick, 2016-07-07:
We will need it submitted in Gerrit, but if he can at least get it cleaned up we can help through that process.

adamliyi, 2016-08-05:
First patch submitted to Gerrit for review. This patch add a fanctl module: https://gerrit.openbmc-project.xyz/#/c/120/
The fanctl module will create a dbus object "/org/openbmc/control/fans". This method will provide these functionalities:
- setMax() set all fans in the system to maximum speed
- updatePresent() update the 'Present' status of fan inventory items
Other method to be added:
- updateFanSpeed() update fan speed based on temperature sensors reading

Also, a fan-monitor service (managed by systemd) will be created. This fan-monitor will call the dbus methods defined in fanctl to manage fans in the system.

closed 2016-10-07 by williamspatrick