openbmc/openbmc #98: Clock management.

williamspatrick, 2015-11-23:
openbmc/phosphor-host-ipmid#40 introduced an implementation of the set-sel-time IPMI function.  Inside it makes a system call to 'hwclock -w' to force write the time to the RTC.

1. Do we want a dbus object for updating the clock(s)?  We discussed possibly having separate BMC and Host clocks as a configuration option.
1. When is the appropriate time to call 'hwclock' and how does is that dealt with if we use ntp.  I know some distros update the hwclock on shutdown.  Can this all be handled via systemd service scripts?

shenki, 2015-11-23:
A third point:

There is a i2c attached DS3231 rtc, which is battery backed (at least on Palmetto), and the internal rtc which is not battery backed.

A normal boot will have the internal RTC at /dev/rtc0, which hwclock -w will default to. The DS3231 will be at /dev/rtc1.

When do we want to sync to/from the battery backed RTC? Do we want to use the internal one at all, if so, it should be disabled in the device tree.

williamspatrick, 2016-06-15:
We should probably break this down into a few different stories.

1. Kernel / device tree work mentioned here by Joel.
2. RTC management for Witherspoon requirements.
3. Larger RTC proposal by @vishwabmc for dual RTC management (BMC vs Host) which is more appropriate for a cloud provider environment.

mdmillerii, 2016-06-15:
1. The rtc was disabled by default in the device tree include in January via openbmc/linux@ed679da11afafa2607bbb45c6f24f7b3d03a9bac.
2. The kernel will read the time at boot
3. The kernel will write the rtc every 11 minutes if ntp is enabled and synchronized.
4. `systemd` will set the kernel timezone from `/etc/adjtime`, although best practice is GMT
5. `systemd` will advance the clock if a stamp in the filesystem is ahead of the kernel time.
6. `systemd` includes a `sntp` client that will sync the time at boot (because they consider date jumps and inaccurate timestamps in journal logs unacceptably broken).

vishwabmc, 2016-07-18:
@shenki : I started to do study and comparison on the time management aspects.. IMO, if we have a RTC  that is not backed by a battery, then we don't want that. We would rather want the RTC be backed  by a battery so that we can count on the RTC ticks when kernel comes up and initializes kernel time / monotonic time. 

BTW, why do we have 2 different RTCs ?.

amboar, 2016-07-18:
@vishwabmc The Aspeed SoCs provide the non-battery-backed RTC IP

geissonator, 2016-11-09:
I assume this is https://gerrit.openbmc-project.xyz/#/c/316/ ?

closed 2016-11-10 by geissonator