openbmc/openbmc #101: Candidate3 driver failing to create a network

causten, 2015-11-25:
root@palmetto:~# uname -a
Linux palmetto 4.3.0-openbmc-20151123-1 #1 Mon Nov 23 13:03:15 CST 2015 armv5tejl GNU/Linux
root@palmetto:~# ifconfig eth0 9.3.164.147 netmask 255.255.255.128
root@palmetto:~# route add default gw 9.3.164.129
root@palmetto:~# cho "nameserver 9.3.1.200" >> /etc/resolv.conf
-sh: cho: command not found
root@palmetto:~# echo "nameserver 9.3.1.200" >> /etc/resolv.conf
root@palmetto:~# ping google.com
^C
root@palmetto:~# ping google.com^C
root@palmetto:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.3.0-openbmc-20151123-1 (causten@causten-VirtualBox) (gcc version 4.9.3 (GCC) ) #1 Mon Nov 23 13:03:15 CST 2015
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Palmetto BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] SOC Rev: 02010303
[    0.000000] On node 0 totalpages: 114176
[    0.000000] free_area_init_node: node 0, pgdat c05149e0, node_mem_map dba79000
[    0.000000]   Normal zone: 892 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114176 pages, LIFO batch:31
[    0.000000] UART IO MUX...
[    0.000000] UART PIN MUX...
[    0.000000] SCU LOCK: 00000001
[    0.000000] DONE, MUX=cb000000 00fff0c0
[    0.000000] CLOCK_CTRL=00000001
[    0.000000] WDT0C=00000000
[    0.000000] WDT2C=00000092
[    0.000000] DLL=1
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113284
[    0.000000] Kernel command line: console=ttyS4,38400
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 430316K/456704K available (3585K kernel code, 171K rwdata, 1268K rodata, 176K init, 109K bss, 26388K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc000000 - 0xff000000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdbe00000   ( 446 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc04c593c   (4855 kB)
                     .init : 0xc04c6000 - 0xc04f2000   ( 176 kB)
                     .data : 0xc04f2000 - 0xc051ccb0   ( 172 kB)
                      .bss : 0xc051ccb0 - 0xc05380bc   ( 110 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Aspeed VIC Initiallized
[    0.000000] clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
[    0.070000] Calibrating delay loop... 190.87 BogoMIPS (lpj=954368)
[    0.070000] pid_max: default: 32768 minimum: 301
[    0.070000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] CPU: Testing write buffer coherency: ok
[    0.070000] Setting up static identity map for 0x40008400 - 0x40008458
[    0.080000] devtmpfs: initialized
[    0.080000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.080000] pinctrl core: initialized pinctrl subsystem
[    0.090000] NET: Registered protocol family 16
[    0.090000] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.160000] SCSI subsystem initialized
[    0.160000] usbcore: registered new interface driver usbfs
[    0.160000] usbcore: registered new interface driver hub
[    0.160000] usbcore: registered new device driver usb
[    0.160000] clocksource: Switched to clocksource moxart_timer
[    0.180000] NET: Registered protocol family 2
[... 172 lines cut ...]
9ec0: c04f8490 c035d978 db674030 db674000 db67400c c035d964 da3a9efc db674000
9ee0: 00000001 c04f8d98 00000000 c04f8d98 da3d0018 db58dac0 db677030 c04f8490
9f00: 00000000 c0027c2c db58dac0 db677030 dba6cc00 db58dac0 db58dad8 db58dad8
9f20: da3a8000 c04f84a0 c04f8490 00000000 00000008 c0027d8c db677034 db58dac0
9f40: c04f8490 c00282ec 00000000 db4593e0 00000000 db58dac0 c0028014 00000000
9f60: 00000000 00000000 00000000 c002bfb8 00000000 00000000 00000000 db58dac0
9f80: 00000000 da3a9f84 da3a9f84 00000000 da3a9f90 da3a9f90 da3a9fac db4593e0
9fa0: c002bef4 00000000 00000000 c000a330 00000000 00000000 00000000 00000000
9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[<c02386d4>] (ftgmac100_hard_start_xmit) from [<c02d2b04>] (dev_hard_start_xmit+0x254/0x2ec)
[<c02d2b04>] (dev_hard_start_xmit) from [<c02e9dac>] (sch_direct_xmit+0x78/0x180)
[<c02e9dac>] (sch_direct_xmit) from [<c02d2f14>] (__dev_queue_xmit+0x288/0x4b0)
[<c02d2f14>] (__dev_queue_xmit) from [<c035b3e0>] (ncsi_xmit_cmd+0x1a8/0x204)
[<c035b3e0>] (ncsi_xmit_cmd) from [<c035cf6c>] (ncsi_dev_start+0x84/0x3a8)
[<c035cf6c>] (ncsi_dev_start) from [<c035d978>] (ncsi_dev_work+0x1a0/0x1ec)
[<c035d978>] (ncsi_dev_work) from [<c0027c2c>] (process_one_work+0x1bc/0x2f8)
[<c0027c2c>] (process_one_work) from [<c0027d8c>] (process_scheduled_works+0x24/0x2c)
[<c0027d8c>] (process_scheduled_works) from [<c00282ec>] (worker_thread+0x2d8/0x3ec)
[<c00282ec>] (worker_thread) from [<c002bfb8>] (kthread+0xc4/0xd8)
[<c002bfb8>] (kthread) from [<c000a330>] (ret_from_fork+0x14/0x24)
Code: e0833201 e1a0ebae e584e47c e2833a01 (e5931000) 
---[ end trace 82c55953c407c3da ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in interrupt

gwshan, 2015-11-25:
I tried to load a private kernel built from mainline. Also I tried to load the kernel by uboot command "run netboot" for couple of times. The problem can't be reproduced. The root cause is the NCSI interface isn't up in time when hitting the kernel crash as below:

- Bring up eth0. NCSI interface isn't up.
- Bring down eth0 and causes kernel crash. It's a known issue I'm tracing currently.

causten, 2015-11-25:
So what happens next?  
 
 Sent from my iPhone using IBM Verse

gwshan, 2015-11-25:
There're two issues here as mentioned: (1) NCSI interface isn't up in time (2) kernel crash

I'm working on a fix for (2). Hopefully I can get something for it. Next step is to recreate (1) then I jump in to debug...

jk-ozlabs, 2015-11-25:
Looks very similar to https://github.com/openbmc/linux/issues/17

shenki, 2015-11-25:
Gavin has posted patches that should resolve the kernel crash. I have merged these patches into the dev-4.3 branch, and tag a release COB today.

causten, 2016-01-12:
I have tried to recreate and it appears the fix is holding.  Thank you

closed 2016-01-12 by causten