openbmc/openbmc #105: ELF relocations in build... bad?

causten, 2015-11-27:
WARNING: QA Issue: ELF binary '/home/causten/openbmc-barreleye/build/tmp/work/armv5e-openbmc-linux-gnueabi/host-ipmid-fru/1.0-r1/packages-split/host-ipmid-fru-dev/usr/lib/libwritefrudata.so' has relocations in .text [textrel]

WARNING: QA Issue: ELF binary '/home/causten/openbmc-barreleye/build/tmp/work/armv5e-openbmc-linux-gnueabi/host-ipmid-fru/1.0-r1/packages-split/host-ipmid-fru/usr/lib/host-ipmid/libstrgfnhandler.so' has relocations in .text [textrel]

WARNING: QA Issue: /usr/lib/host-ipmid/libstrgfnhandler.so_host-ipmid-fru contained in package host-ipmid-fru requires libwritefrudata.so, but no providers found in its RDEPENDS [file-rdeps]

WARNING: QA Issue: ELF binary '/home/causten/openbmc-barreleye/build/tmp/work/armv5e-openbmc-linux-gnueabi/host-ipmid-oem/1.0-r1/packages-split/host-ipmid-oem/usr/lib/host-ipmid/liboemhandler.so' has relocations in .text [textrel]

shenki, 2015-12-17:
I also see this when building the latest:

```
WARNING: QA Issue: /usr/sbin/control@underscore@bmc@underscore@barreleye.exe_skeleton contained in package skeleton requires libopenbmc_intf.so, but no providers found in its RDEPENDS [file-rdeps]
```

causten, 2016-02-19:
fixed

closed 2016-02-19 by causten