openbmc/openbmc #107: linux: WARN on boot in networking code

shenki, 2015-12-01:
Using an OpenBMC userspace and a 4.3 kernel with some unrelated watchdog debugging patches.

```
WARNING: CPU: 0 PID: 422 at kernel/softirq.c:150 __local_bh_enable_ip+0x48/0xbc()
Modules linked in:
CPU: 0 PID: 422 Comm: systemd-journal Not tainted 4.3.0-00027-g6041e18256bd-dirty #175
Hardware name: ASpeed SoC
[<c000f288>] (unwind_backtrace) from [<c000cf18>] (show_stack+0x10/0x14)
[<c000cf18>] (show_stack) from [<c0016734>] (warn_slowpath_common+0x84/0xac)
[<c0016734>] (warn_slowpath_common) from [<c00167ec>] (warn_slowpath_null+0x18/0x20)
[<c00167ec>] (warn_slowpath_null) from [<c0018ef4>] (__local_bh_enable_ip+0x48/0xbc)
[<c0018ef4>] (__local_bh_enable_ip) from [<c02cb0ec>] (dst_alloc+0x50/0x84)
[<c02cb0ec>] (dst_alloc) from [<c02e4894>] (rt_dst_alloc+0x4c/0x58)
[<c02e4894>] (rt_dst_alloc) from [<c02e6154>] (ip_route_input_noref+0x608/0x7d8)
[<c02e6154>] (ip_route_input_noref) from [<c03121fc>] (arp_process+0x22c/0x668)
[<c03121fc>] (arp_process) from [<c0312968>] (arp_rcv+0x108/0x17c)
[<c0312968>] (arp_rcv) from [<c02c01cc>] (__netif_receive_skb_core+0x668/0x79c)
[<c02c01cc>] (__netif_receive_skb_core) from [<c02c17b8>] (netif_receive_skb_internal+0x4c/0x58)
[<c02c17b8>] (netif_receive_skb_internal) from [<c02c1f54>] (napi_gro_receive+0x48/0x9c)
[<c02c1f54>] (napi_gro_receive) from [<c022a4e4>] (ftgmac100_poll+0x39c/0x58c)
[<c022a4e4>] (ftgmac100_poll) from [<c02c23f0>] (net_rx_action+0xd4/0x28c)
[<c02c23f0>] (net_rx_action) from [<c0018cbc>] (__do_softirq+0xcc/0x21c)
[<c0018cbc>] (__do_softirq) from [<c003af84>] (__handle_domain_irq+0x84/0xa0)
[<c003af84>] (__handle_domain_irq) from [<c00093d4>] (avic_handle_irq+0x5c/0x64)
[<c00093d4>] (avic_handle_irq) from [<c000d970>] (__irq_svc+0x50/0x64)
Exception stack(0xc81c5e18 to 0xc81c5e60)
5e00:                                                       c818f240 00000000
5e20: cf78b46e c00d0160 cf78c070 cf78b440 c81c5ea8 c81c5ea0 cfab5330 c81c5ed8
5e40: 00000001 be9b63ec 00000065 c81c5e68 c0094954 c00d0174 40000013 ffffffff
[<c000d970>] (__irq_svc) from [<c00d0174>] (pid_revalidate+0x14/0xb8)
[<c00d0174>] (pid_revalidate) from [<c0094954>] (lookup_fast+0x200/0x284)
[<c0094954>] (lookup_fast) from [<c0094d78>] (walk_component+0xb0/0x290)
[<c0094d78>] (walk_component) from [<c009631c>] (path_lookupat+0x80/0xfc)
[<c009631c>] (path_lookupat) from [<c0098304>] (filename_lookup+0x5c/0xb0)
[<c0098304>] (filename_lookup) from [<c00903f4>] (SyS_readlinkat+0x4c/0xd0)
[<c00903f4>] (SyS_readlinkat) from [<c000a280>] (ret_fast_syscall+0x0/0x38)
---[ end trace 077cd510230430b6 ]---
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x257: NO_RXBUF 
ftgmac100: NCSI dev is up
```

shenki, 2015-12-01:
@gwshan to take a look

gwshan, 2015-12-03:
It's a warning message printed when enabling bottom half in softIRQ path and it's not critical issue. The issue isn't reproducible every time. I might hit one in continuous 10+ times. I'm travelling now and will look for more after settlement.

williamspatrick, 2016-01-09:
@gwshan - Is this resolved?

nkskjames, 2016-03-03:
this is fixed

closed 2016-03-03 by nkskjames