openbmc/openbmc #122: SDK build fails on ppc64le

agangidi53, 2015-12-15:
While configuring libgcc SDK build fails when run on ppc64le. I tried this on Barreleye with Ubuntu 15.04 .

shenki, 2015-12-16:
Can you please provide the logs for your failed build?

agangidi53, 2015-12-17:
Hi Shenki 

@causten  should have the logs. I did this build live image and lost access to it. Ii'll re-run the build if he doesn't have them.

anoo1, 2015-12-17:
$ bitbake -v -c populate_sdk obmc-phosphor-image
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    libsdl-native is set to be ASSUME_PROVIDED but sdl-config can't be found in PATH. Please either install it, or configure qemu not to require sdl.

Summary: There was 1 ERROR message shown, returning a non-zero exit code.

$ sudo apt-get install libsdl1.2-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libsdl1.2-dev
E: Couldn't find any package by regex 'libsdl1.2-dev'

---
Doing a quick web search, didn't find an installable libsdl1.2 deb package for ppc64 little endian.

shenki, 2015-12-17:
That's strange.

As a workaround, you can build Qemu without SDL. Find `build/conf/local.conf` and comment out the this block:

```
# By default qemu will build with a builtin VNC server where graphical output can be
# seen. The two lines below enable the SDL backend too. This assumes there is a
# libsdl library available on your build system.
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
ASSUME_PROVIDED += "libsdl-native"
```

shenki, 2015-12-17:
```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="14.04.3 LTS, Trusty Tahr"
```
```
$ apt-cache policy libsdl1.2-dev 
libsdl1.2-dev:
  Installed: (none)
  Candidate: 1.2.15-8ubuntu1.1
  Version table:
     1.2.15-8ubuntu1.1 0
        500 http://ports.ubuntu.com/ubuntu-ports/ trusty-updates/main ppc64el Packages
```

closed 2016-03-03 by nkskjames