openbmc/openbmc #124: os-release is not updated after change to git repository

bradbishop, 2015-12-18:
I'm working on a way to make this get regenerated whenever something changes in git.

also..I used git tag where I should have used git describe.

bradbishop, 2015-12-18:
I have https://github.com/openbmc/openbmc/pull/125 that addresses this in a hacky way.  Planning on leaving this open until I can figure out how to make the package rebuild only after a git repository change.

eddiejames, 2016-09-08:
I set up the os-release bbappend to compile only when there are changes to the git repository. This works the first time you run the recipe after it has been modified. 

The problem is that bitbake appears to cache the recipe and will no longer accept changes to the task list (compile, install, etc) after it is parsed for the first time after the recipe file is modified. For example, even if the recipe is parsed and sets do_compile[noexec] = "1", it makes no difference and compile is executed anyway if it was executed the first time the recipe was parsed. Even if I do a cleanall, if the recipe didn't compile at first, it will still not compile.

I'm not aware of any other ways to control the tasks other than using the noexec or nostamp tags to respectively disable or enable a task. I have tried setting these in a variety of tasks, events, and just within the recipe, but all the results are the same.

rfrandse, 2017-07-25:
<https://gerrit.openbmc-project.xyz/5466> os-release: Don't rebuild everytime
Resolves:  #124 os-release is not updated after change to git repository

closed 2017-08-01 by williamspatrick