openbmc/openbmc #128: palmetto build still tries to build qemu

jk-ozlabs, 2016-01-07:
I've configured openbmc for the palmetto target:

    $ TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-palmetto/conf . oe-init-build-env

Resulting in:

    $ grep -R ^MACHINE conf
    conf/local.conf:MACHINE ??= "palmetto"

However, build is failing during qemu do_compile. While the actual build failure (of `qemu-native`) is a problem, can we just skip this step for non-qemu platforms?

jk-ozlabs, 2016-01-07:
Also, qemu seems to be the package that is most sensitive to build-system differences, so this would prevent build problems on varied build systems...

anoo1, 2016-01-08:
If you've built qemu before, you might need to delete the conf dir: rm -rf build/conf  ..  then try the export and bitbake commands once more.

jk-ozlabs, 2016-03-10:
@anoo1: this is a fresh build

williamspatrick, 2016-04-01:
Adding qemu for palmetto was being worked.  I think the real issue is that qemu-native should not fail to compile when the host machine does not have SDL.  It should detect this and degrade to an SDL-less compile.

williamspatrick, 2016-06-15:
1. Need to ensure that qemu is not a dependency for all machine configurations.
2. Need to ensure that qemu-native compiles for qemuarm / qemux86 targets on something like Ubuntu Server Edition.

closed 2019-10-27 by stale[bot]