openbmc/openbmc #132: Fails to boot host on palmetto

shenki, 2016-01-13:
I'm running the latest pnor and openbmc builds:

OpenBMC: https://github.com/openbmc/openbmc/commit/df37619ce77c8b7f6805b8bc319ec15abfcdf9dc
PNOR: https://github.com/open-power/op-build/commit/da2732d9bde6351c4668f50a63c9a02aeaec30d6

OpenBMC boots fine. I then run the following to start the host:

```
root@palmetto:~# obmcutil poweron
 = None
```

And nothing happens on the host serial port.

```
root@palmetto:~# obmcutil power  
pgood = 1
state = 1
pgood_timeout = 10
```

anoo1, 2016-01-13:
Power on is failing because the Throttle gpio was added only to Barreleye and not Palmetto. This is now fixed in openbmc commit 23ffc6d. Just need to refresh your openbmc repo and rebuild the image.

root@palmetto:~# obmcutil poweron
 = None
root@palmetto:~# journalctl --follow
-- Logs begin at Sun 2015-12-13 21:19:41 UTC. --
Jan 13 19:31:23 palmetto system_manager.py[797]: Turn on power and boot
Jan 13 19:31:23 palmetto system_manager.py[797]: Running System State: HOST_POWERING_ON
Jan 13 19:31:23 palmetto system_manager.py[797]: Setting watchdog poll interval to: 30000
Jan 13 19:31:23 palmetto system_manager.py[797]: Starting watchdog with poll interval: 30000
Jan 13 19:31:24 palmetto system_manager.py[797]: Running System State: HOST_POWERED_ON
Jan 13 19:31:24 palmetto system_manager.py[797]: Booting host
Jan 13 19:31:24 palmetto system_manager.py[797]: ERROR HostControl: GPIO sequence failed (rc=2)

shenki, 2016-01-14:
Thanks for the quick response @anoo1.

I updated to https://github.com/openbmc/openbmc/commit/23ffc6d9d39c0b48fafb8ac4f1efe65611a27922 (Build 35 https://openpower.xyz/job/openbmc-build/35/distro=ubuntu,target=palmetto/) and I still cannot boot my host. Are you able to test the image that our build system has produced?

anoo1, 2016-01-14:
We've narrowed down the issue to the op-build master-next image. It doesn't boot on palmetto with openbmc. But the op-build master image (most recent is from oct 2015) boots ok with the latest openbmc. Working with hostboot team to identify what change that was dropped between oct and now that may be causing the issue.

mkumatag, 2016-01-30:
Even I tried booting up the machine in one of the palmetto with build 42 - https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=palmetto/42/ but couldn't boot up and seen following messages in journalctl --follow command:

Jan 30 10:07:35 palmetto system_manager.py[824]: Turn on power and boot
Jan 30 10:07:35 palmetto system_manager.py[824]: Running System State: HOST_POWERING_ON
Jan 30 10:07:35 palmetto system_manager.py[824]: Setting watchdog poll interval to: 30000
Jan 30 10:07:35 palmetto system_manager.py[824]: Starting watchdog with poll interval: 30000
Jan 30 10:07:36 palmetto system_manager.py[824]: Running System State: HOST_POWERED_ON
Jan 30 10:07:36 palmetto system_manager.py[824]: Booting host
Jan 30 10:07:36 palmetto system_manager.py[824]: Using primary side of the bios flash
Jan 30 10:07:36 palmetto system_manager.py[824]: Running System State: HOST_BOOTING
Jan 30 10:08:05 palmetto system_manager.py[824]: Polling watchdog: 1
Jan 30 10:08:34 palmetto systemd[1]: Starting Cleanup of Temporary Directories...
Jan 30 10:08:34 palmetto systemd-tmpfiles[977]: Failed to open directory /var/tmp: Too many levels of symbolic links
Jan 30 10:08:34 palmetto systemd[1]: Started Cleanup of Temporary Directories.
Jan 30 10:08:35 palmetto system_manager.py[824]: Polling watchdog: 0
Jan 30 10:08:35 palmetto system_manager.py[824]: Watchdog Error, Hard Rebooting

No activity seen in the host console of the system.

anoo1, 2016-02-18:
@williamspatrick , this is the issue that hostboot team is asking if you could help.

anoo1, 2016-02-19:
The issue is not seeing anymore now that openbmc added vuart support. I'll open a hostboot issue so they can follow-up for systems that do not have vuart and use hw instead.

closed 2016-03-03 by nkskjames