openbmc/openbmc #135: wget does not work

shenki, 2016-01-14:
In our lab, test machines are firewalled from the web and must go through a http proxy:

```
root@palmetto:~# wget https://openpower.xyz/job/openbmc-build/lastSuccessfulBuild/distro=ubuntu,target=palmetto/artifact/images/images/palmetto/flash-palmetto
Connecting to openpower.xyz (104.239.249.49:443)
wget: error getting response: Connection reset by peer
```

Setting the proxy still fails:

```
root@palmetto:~# export http_proxy=proxy:3128root@palmetto:~# wget https://openpower.xyz/job/openbmc-build/lastSuccessfulBuild/distro=ubuntu,target=palmetto/artifact/images/images/palmetto/flash-palmetto
Connecting to proxy:3128 (10.61.2.1:3128)
wget: error getting response: Connection reset by peer
```

jk-ozlabs, 2016-01-14:
How about:

    export https_proxy=http://proxy:3128/

?

shenki, 2016-01-14:
@jk-ozlabs oops, that's what I was doing, but I copied it incorrectly into this issue.

Obviously setting http_proxy for a HTTPS connection has no effect.

nkskjames, 2016-03-03:
marking as an enhancement to enable https proxy.

mdmillerii, 2016-03-17:
busybox wget today only looks for http_proxy and ftp_proxy and calls out to openssl for HTTPS.

openssl just added a -proxy option in May and it isn't in our current version.  We can check again after our next yocto update, and update openssl or backport the patch.  I didn't find any update for busybox to use the new functionality yet either, but that should be straight forward from my initial look.

Base https://user:password@host/ has been verified to work with the obvious security concerns of the URL including the password appearing on the process command line.

williamspatrick, 2016-04-01:
I put this in the v2.0 backlog because we will be sticking with yocto-2.0.1 for the v1.0 release.  2.0.1 is on the OpenSSL 1.0.2 stream and this feature was not added until 1.1.

```
$ git log -n1 552bf8ec | cat
commit 552bf8ec5e64d1a169069111850ebc5d250e0499
Author: Michael Trapp <m.trapp@gmx.de>
Date:   Fri May 8 15:34:07 2015 -0400

    RT266: Add HTTP proxy/CONNECT to s_client
    
    Signed-off-by: Rich Salz <rsalz@openssl.org>
    Reviewed-by: Richard Levitte <levitte@openssl.org>
$ git tag --contains 552bf8ec5e64d1a169069111850ebc5d250e0499
OpenSSL_1_1_0-pre1
OpenSSL_1_1_0-pre2
OpenSSL_1_1_0-pre3
OpenSSL_1_1_0-pre4
```

williamspatrick, 2016-06-23:
Yocto 2.1 still uses openssl-1.0.2g, so still not going to get a fix for this.  Actually, even Yocto is waiting on openssl to even do a formal release of the 1.1.x stream.

gtmills, 2018-07-09:
@shenki Looks like Yocto 2.5 came with a openssl_1.1.0h.bb,  https://github.com/openbmc/openbmc/blob/master/import-layers/yocto-poky/meta/recipes-connectivity/openssl/openssl_1.1.0h.bb

closed 2019-10-28 by stale[bot]