openbmc/openbmc #137: Verify mac addresses are set correctly

shenki, 2016-01-18:
We currently rely on the uboot environment variable to store the mac address. Both uboot and Linux should use this information; verify that this is the case and ensure we can detect when it fails.

jk-ozlabs, 2016-01-18:
.. also, this means that a system implementer will need to set per-machine u-boot variables during manufacturing. We'll need to ensure that this is acceptable, or find somewhere else (another EEPROM?) to store the MAC address.

nkskjames, 2016-01-18:
The plan was for the manufactuer to use the IPMI set LAN from host to set MAC address.  This should set both MACs.  Once we have the fw flash tools in Linux, we can use to set uboot variables.

agangidi, 2016-01-29:
I verified today on my 24 servers that the mac address you set via uboot is persistent in both uboot and linux. However as Norm said, setting it via linux / host OS is a better idea for future purposes.

nkskjames, 2016-03-03:
Marking as an enhancement to store MAC in EEPROM and copy into uboot environment.

anoo1, 2016-09-26:
Available in:
Master - https://github.com/openbmc/skeleton/commit/fb4d22587fdc46d64c253a1ab1b40b228a8c0887
Stable - https://github.com/openbmc/skeleton/commit/6019000c3cc52d33d85906c93ffd1b8614b65c59

closed 2016-09-26 by anoo1