openbmc/openbmc #147: obmc-rest is using more than 40% of CPU resources

shenki, 2016-01-27:
While running a recent build of openbmc I noticed the following:

```
  912     1 root     S     113m  46%   0% python /usr/sbin/obmc-rest
```

bradbishop, 2016-01-27:
Are you sure that's not memory?  I have a pull request to drop the number of server threads that helped some.

shenki, 2016-01-27:
Ah yep, you're right.

That's lots of memory. Do we expect it to use that much?

bradbishop, 2016-01-27:
Yeah.  With less threads (2 I think it was) it went down to about 17%, which still seems high.  Rocket is pretty fast - we can probably trade up for something slower that uses less memory.  The WSGI servers are just drop in replacements and there are quite a few to pick from.

shenki, 2016-01-27:
Thanks for the explanation, I'll leave the bug open as a reminder.

KennethWilke, 2016-02-12:
This is a good comparison of wsgi servers, don't think they measure rocket but it might help anyway: http://nichol.as/benchmark-of-python-web-servers

I recommend [uWSGI](https://uwsgi-docs.readthedocs.org/en/latest/) though, it's pretty fast and light

bradbishop, 2016-08-21:
Hello, 

I've seen a couple of things that might interest you, just take a look here <http://bandebywe.apachepeakconcepts.com/e4fjt>

Sincerely, bradleyb

bradbishop, 2016-08-21:
Hello, 

I've  found some fresh  and cool  stuff, don't miss it up!  More info  here <http://crolufenki.frankparry.net/e4hyxo>

Warmly, bradleyb

hweaving, 2016-08-29:
I prototyped adding nginx + gunicorn.  Running nginx itself seems lightweight and nice to have in general as a proxy server + static content server.

However, gunicorn seemed to take a lot of flash and RAM.  I agree with uWSGI, and/or gevent.  It was suggested that a PREFERRED_PROVIDER bitbake var could choose between uWSGI and gevent.

bradbishop, 2016-09-08:
I've added python-gevent and set it as the default for phosphor.  I've also added enough plumbing to facilitate others wanting to try alternatives with something like:

PREFERRED_PROVIDER_obmc_wsgi_host = "my-uwsgi-andor-gunicorn-andor-nginx-packagegroup"

in another layer or from local.conf.

You would just need to then implement some kind of startup script/application for the alternative server and point it at /etc/default/obmc/wsgi_app.

As of today the patchset is still under review and hasn't been mainlined yet.

williamspatrick, 2016-09-09:
Code is all merged into master.

closed 2016-09-09 by williamspatrick