openbmc/openbmc #184: ftgmac hitting errors

causten, 2016-02-18:
I let the BMC sit for >10 minutes with the host in a hung state.  These prints just started popping up

-rw-r--r--    1 root     root           154 Feb 18 16:33 9
root@palmetto:~# ftgmac100 1e660000.ethernet eth0: [ISR] = 0x207: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
net_ratelimit: 2 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x207: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST

shenki, 2016-02-18:
Do you have the output of `dmesg`?

What is the kernel version?

causten, 2016-02-18:
root@palmetto:~# uname -r
4.3.5-openbmc-20160212-1

dmesg is full of the prints.  maybe it wrapped?

williamspatrick, 2016-06-15:
Is this fixed now?

mdmillerii, 2016-06-15:
The error is the ethernet driver saying it threw away packets because it was not serviced and had no space remaining.  There have been several changes to the kernel including locking changes in this ethernet driver and memory corruption fixes since this was reported.

Is this recreatable?   Was it ever?

shenki, 2016-06-15:
I've not seen this. Let's close it.

geissonator, 2016-06-29:
palm303 suddenly stopped pinging, I went out to the lab and verified the connections (all was good).  The serial looked fine and the IP settings looked fine so I issued a reboot.  When it came up I got at lot of these NO_RXBUF RPKT_LOST errors so looks like this same issue.  I'm thinking I'll try a A/C power cycle next.

ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
root@palmetto:~#  cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0-11-gf81616d-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0

root@palmetto:~# uname -a
Linux palmetto 4.4.12-openbmc-20160606-1 #1 Fri Jun 10 22:13:13 UTC 2016 armv5tejl GNU/Linux

geissonator, 2016-06-29:
An A/C power cycle fixed the issue.  Interesting that just a "reboot" didn't.

mdmillerii, 2016-06-29:
@geissonator Can you check if the mac address is set in the `u-boot` environment  or if its using a random mac?  Either `printenv ethaddr` from the `u-boot` prompt or `fw_printenv ethaddr` from Linux prompt (or just `printenv` or `fw_printenv` respectively and look for the `ethaddr` variable.

The fact that reboot did not clear but power cycle did could be a poorly chosen random mac that the switch stopped liking, or it could be the code to reset and initialize the adapter did not handle some prior state.  On a simple reboot if the mac is retained in the hardware registers, but on a power cycle the register is zeroed and `u-boot` assigns a random mac (with 2 bits fixed per IEEE rules).

geissonator, 2016-06-29:
The problem came back after a bit of testing :(  

Here's my current mac with system in fail state:
fw_printenv ethaddr
ethaddr=00:21:cc:73:91:dd

geissonator, 2016-06-29:
I moved to a different switch and have seen no issues so far.

williamspatrick, 2016-06-29:
@shenki - I'll leave this open for you to assess if there is something for us to do here.  Concerning to me that a flaky or busy switch could give us trouble.

shenki, 2016-06-30:
From IRC:

```
12:31 < miltonm> AndewG_ https://lists.ozlabs.org/pipermail/openbmc/2016-June/004032.html looks intresting  
12:32 < miltonm> AndrewG_ ^^  
12:34 < AndrewG_> that was quick :)  
12:35 < miltonm> AndrewG_ but it seems to say something is flooding your network ... since its happening at the kernel it might just be junk ? like pause packets?  
12:37 < AndrewG_> Def not my area of expertise.  The switch it's plugged into is fully populated and very busy.  
12:37 < miltonm> I guess we would need to see driver stat counters (ifconfig  output)  
12:37 < miltonm> AndrewG_ I just remember debugging problems on other adapters ... yea  
12:38 < AndrewG_> RX packets:49927286 errors:0 dropped:49754196 overruns:1045 frame:1151  
12:39 < miltonm> AndrewG_ that's a lot of packets and its not just (u32)-1  
12:45 < miltonm> about 35x another box in the same rack  
```

shenki, 2016-08-30:
Has this been reproduced with the latest stable kernel version (openbmc-4.4-20160819-1)?

geissonator, 2016-09-01:
This showed up on palm205 today.  Leading up to it, the IP would not ping, so I was just going about resetting all the different IP parameters (even though they were all correct already)

I'll leave system reserved and in fail state in case someone wants to look at it (serial is working fine).

ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
root@palmetto:~# cat /etc/version && cat /etc/os-release && uname -a
20160831020951
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.0-174"
VERSION_ID="v1.0-174-ga81b2a7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.0-174"
BUILD_ID="v1.0"
Linux palmetto 4.7.1-6d93f625315a1e9b51d017a78f8c0938a097ea85 #1 Wed Aug 17 10:18:14 UTC 2016 armv5tejl GNU/Linux

          RX packets:2261876 errors:0 dropped:2261877 overruns:273 frame:311
          TX packets:32 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:113093800 (107.8 MiB)  TX bytes:2136 (2.0 KiB)

shenki, 2016-09-02:
Note that @geissonator is running a different kernel version than the original bug report.

@gwshan, any ideas here?

williamspatrick, 2016-09-19:
```
root@palmetto:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-85"
VERSION_ID="v1.99.0-85-g0d2bd6a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-85"
BUILD_ID="v1.99.0"
root@palmetto:~# uname -a
Linux palmetto 4.7.1-6d93f625315a1e9b51d017a78f8c0938a097ea85 #1 Wed Aug 17 10:18:14 UTC 2016 armv5tejl GNU/Linux
```

mdmillerii, 2016-09-19:
from `ifconfig eth0` on palm205 this morning, RX packes and dropped match:

```
ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 00:21:CC:73:91:DD  
          inet addr:9.3.164.177  Bcast:9.3.164.255  Mask:255.255.255.128
          inet6 addr: fe80::221:ccff:fe73:91dd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:142640487 errors:0 dropped:142640487 overruns:1860 frame:1986
          TX packets:1874 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2837057054 (2.6 GiB)  TX bytes:79500 (77.6 KiB)
```

I would suggest a block of errors is displayed every few minutes, the first with status `0x5f` the rest with `0xf` and 2 lines suppressed `with net_ratelimit`
```
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
net_ratelimit: 2 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
```

gwshan, 2016-09-20:
It's likely flooding attack caused by broadcast packets. Too many packets are received and they don't need responses according to the Rx/Tx statistics showed by @mdmillerii

Do we have idea on how to reproduce it? I need recreate it here first of all.

gwshan, 2016-09-20:
Thanks to Patrick for machine sharing. I think we have to figure out another issue before coming to the broadcast packet attack issue: It seems the NCSI link was bouncing and at that point, I didn't see any packet was dropped from "ifconfig eth0" as below.

ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
random: nonblocking pool is initialized
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up

root@palmetto:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 00:21:CC:73:91:DD  
          inet addr:9.3.164.177  Bcast:9.3.164.255  Mask:255.255.255.128
          inet6 addr: fe80::221:ccff:fe73:91dd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:63048 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1321 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3389906 (3.2 MiB)  TX bytes:85401 (83.3 KiB)

I believe the link down/up was reported by NCSI AEN packet as the interval is very small between the link-down and link-up event. Also, I loaded my own private kernel (4.7+buildroot initramfs) and didn't see this issue there. It seems some one (BMC application) or host is causing this conflicts.

The broadcast packet attack issue would be related to this. The current code doesn't handle above scenario very well, leading to corrupted NCSI MAC filter configuration eventually. I will try below experiments on local machine to validate: (A) Configure same MAC address to two channels at same time; (B) Configure multiple filters with same MAC address on active NCSI channel.

geissonator, 2016-09-21:
Looks like our palm205 hit this over night here in Austin.  Let me know if there's anything you'd like collected in the fail state

ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x20f: NO_RXBUF RPKT_LOST 
root@palmetto:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr A0:42:3F:25:17:1E  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:104279572 errors:0 dropped:100517251 overruns:906 frame:996
          TX packets:1141 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:930277323 (887.1 MiB)  TX bytes:188557 (184.1 KiB)

williamspatrick, 2016-09-21:
An added note on palm205 is there are no up/down messages now.  Just the NO_RXBUF RPKT_LOST messages.  So that rules out PCI reset in my mind.

gwshan, 2016-09-21:
@williamspatrick, Please check if there are link down/up messages in last boot.

williamspatrick, 2016-09-21:
@gwshan - I don't understand what you're asking.  Every time we boot the host there will be a down/up sequence.  That is expected.

@geissonator - I just noticed we [you] left the machine running stable instead of master.  We need to put a 4.7.3 kernel image onto this because @gwshan felt there were some fixes there.

gwshan, 2016-10-04:
Please update if we still hit this issue after resolving the same MAC address assigned to two BMCs which are resident in same LAN.

geissonator, 2016-10-04:
Hey Gavin, fixing the mac address issue and ensuring we didn't run test cases that changed the mac address definitely fixed a lot of our issues.  We continue to occasionally see situations where our barreleyes get the down/up message and then no longer ping.  A power cycle of the system fixes the issue (I've seen it twice in the last week so very intermittent).

palm205 (our problem child palmetto) still is spewing errors but our other palmetto has been fine.

```
net_ratelimit: 2 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST 
ifconfig
eth0      Link encap:Ethernet  HWaddr A0:42:3F:25:17:1E  
          inet6 addr: fe80::a242:3fff:fe25:171e%984/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:828045035 errors:0 dropped:828045035 overruns:7478 frame:8033
          TX packets:6592 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2747546086 (2.5 GiB)  TX bytes:2180056 (2.0 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%984/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1101536 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1101536 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:83716704 (79.8 MiB)  TX bytes:83716704 (79.8 MiB)

root@palmetto:/etc/systemd# uname -a
Linux palmetto 4.7.3-be2001133835be80ba4657232b823ba2daaac0d6 #1 Wed Sep 14 12:05:02 UTC 2016 armv5tejl GNU/Linux
```

gwshan, 2016-10-07:
@geissonator, thanks for sharing the BMC for me to reproduce the issue. Unfortunately, I cannot recreate it on palm 205. Joel has pushed couple of NCSI fixes to last 4.7 kernel. Lets see if any one of those NCSI patches resolves this issue.

However, there is one experiment I will do as Joel suggested: connect laptop and BMC directly and write up a program to send ARP request or ICMP request to BMC without limitation, to see if it can trigger the issue... will update the result when it's done.

geissonator, 2016-10-12:
Put the latest kernel on BarPVT

Saw same issue I'd pinged Gavin on a while ago.  As soon as I issue the power on, the network drops and does not come back.

We get this in the console:
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up

Doing this did not help:
root@barreleye:~#  ifconfig eth0 down; ifconfig eth0 up; echo $?
ftgmac100 1e660000.ethernet eth0: NCSI interface down
0
root@barreleye:~# ftgmac100 1e660000.ethernet eth0: NCSI interface up

root@barreleye:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-170"
VERSION_ID="v1.99.0-170-gf735c59-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-170"
BUILD_ID="v1.99.0"
root@barreleye:~# uname -a
Linux barreleye 4.7.7-9fc87f9c1103ad94063f2a819f21117c166e059d #1 Wed Oct 12 04:17:59 UTC 2016 armv5tejl GNU/Linux

geissonator, 2016-10-12:
@gwshan I'm going to leave BarPVT in the fail state.  I did a 2nd PDU power cycle, got network back, issued the power on and once again the network dropped on me.  You should have all the access info for the system but I'll also be online tonight if needed.

gwshan, 2016-10-12:
Andrew, thanks for sharing the machine for debugging. I think the BCM57xx
was stuck. After executing below command to reset it, the network is coming
back.

root@barreleye# gpioutil -n "PCIE_RESET" -v 1
GPIO ID: 333; DIRECTION: out
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up

root@barreleye# gpioutil -n "PCIE_RESET"
GPIO ID: 333; DIRECTION: out
0

root@barreleye# ping 9.3.23.1
PING 9.3.23.1 (9.3.23.1): 56 data bytes
64 bytes from 9.3.23.1: seq=0 ttl=64 time=4.192 ms
64 bytes from 9.3.23.1: seq=1 ttl=64 time=2.079 ms

gwshan, 2016-10-13:
I assume the issue happens only after giving a power-cycle to BMC. I used below script doing host power cycle for 200 times. I cannot hit the issue:

let i=0; \
while true; \
   do echo "[$i] Power cycle"; \
   obmcutil poweroff; sleep 1; \
   obmcutil poweron; sleep 5; \
   ping -c 1 -w 3 9.3.23.1; \
   if [ $? -ne 0 ]; then break; fi; \
   i=$[$i+1]; done

[206] Power cycle
 = None
 = None
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
PING 9.3.23.1 (9.3.23.1): 56 data bytes
64 bytes from 9.3.23.1: seq=0 ttl=64 time=0.977 ms

--- 9.3.23.1 ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 0.977/0.977/0.977 ms

gwshan, 2016-10-13:
The issue is 100% reproducible on this barreleye machine. I believe two ethernet cables are connected on this machine, @geissonator, please help to confirm. I built private kernel from dev-4.7 and use below commands to do power host on/off. The observed AEN packets are as below:

# GPIO E1 for host's power, high level sensitive
devmem 0x1e780024 32 0x00000002
devmem 0x1e780020 32 0xFFC8F71D         # Power-on
devmem 0x1e780020 32 0xFFC8F71F         # Power-off

Event                Channel#0                 Channel#1                 Index
=======================================================
Power-on          Down                         Up                              1
Power-off                                            Up                              2
Power-on          Up                             Down                          3
Power-off          Down                                                           4
Power-on          Up                                                  <--- Since then, LSC AEN for #0 is received only
Power-off          Down

What is happened behind and everything matches with the code:
(1) Channel#0 is down and switch over to Channel#1. However, it seems Channel#1 is connected to another LAN, meaning the currently assigned IP address doesn't work there.
(3) Channel#1 is down and switch over to Channel#0. NCSI network works again.
(..) Channel#0 is always the active channel, we cannot hit the issue in subsequent host power on/off.

I already had a patch that is included in the private kernel (9.3.164.219:/tftpboot/uImage.gavin). With the kernel booted on the machine, everything looks fine. I'll push the fix upstream and backport to dev-4.7 afterwards. Thanks to Andrew for sharing machine for debugging again:

- The system boots up. Network works fine:
# ping -c 1 9.3.23.1
PING 9.3.23.1 (9.3.23.1): 56 data bytes
64 bytes from 9.3.23.1: seq=0 ttl=64 time=4.963 ms

- Power on the host, everything looks good:
# NCSI: AEN-LSC (0,0): up -> down
ftgmac100 1e660000.ethernet eth0: NCSI interface down
NCSI: AEN-LSC (0,0): down -> up
ftgmac100 1e660000.ethernet eth0: NCSI interface up
# ping -c 1 9.3.23.1
PING 9.3.23.1 (9.3.23.1): 56 data bytes
64 bytes from 9.3.23.1: seq=0 ttl=64 time=1.598 ms

- Power on the host again, network works fine:
# devmem 0x1e780020 32 0xFFC8F71F
# NCSI: AEN-LSC (0,0): up -> up
# devmem 0x1e780020 32 0xFFC8F71D
# NCSI: AEN-LSC (0,0): up -> down
ftgmac100 1e660000.ethernet eth0: NCSI interface down
NCSI: AEN-LSC (0,0): down -> up
ftgmac100 1e660000.ethernet eth0: NCSI interface up
# ping -c 1 9.3.23.1
PING 9.3.23.1 (9.3.23.1): 56 data bytes
64 bytes from 9.3.23.1: seq=0 ttl=64 time=1.685 ms

geissonator, 2016-10-13:
Yes, 2 enet cables are plugged into this system.

gwshan, 2016-10-13:
@geissonator, thanks for confirm. I also believe the IP addresses assigned to those two ethernet ports (BMC and OS) are in different LAN, they cannot talk to each other directly. So far, the issue (initially caused by BCM57xx's strange behaviour) and fix are clear. Please wait for the fix to show in dev-4.7 branch. For now, the workaround is: you have to issue two host power-off/on cycle. After that, everything should be back to normal track.

shenki, 2016-10-18:
Gavin has submitted a series upstream for review. Once he sends v2, we can backport those changes to our dev-4.7 tree.

https://patchwork.ozlabs.org/patch/682097/
https://patchwork.ozlabs.org/patch/682098/
https://patchwork.ozlabs.org/patch/682099/
https://patchwork.ozlabs.org/patch/682101/
https://patchwork.ozlabs.org/patch/682100/

gwshan, 2016-10-20:
Couple of patches that have been put into Dave's git repo ("net" branch) and should reach upstream shortly. I also backported and posted them for our dev-4.7 branch. For the particular issue (network lost after power host on) should be fixed by below one:

https://patchwork.ozlabs.org/patch/684803/

If possible, please close this one and create new bug when having further issue with NCSI stack. It's always nice have one bug tracking one particular issue, thanks!

closed 2016-10-21 by williamspatrick

gwshan, 2017-02-24:
I was writing a kernel driver to operate the GPIO PINs so that I can keep powering on/off the host while running application to accept TCP connections. All of this is for another bug reported from Foxconn. The driver doesn't work well yet and helps recreating the issue. On Palmetto, I have last 4.7 branch running and eventually see below messages.

ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST

Then I reboot into uboot and try load the kernel image via NCSI channel again, but it doesn't work and keep outputting messages like below. Intrestingly, do we have a PHY and how its address can be 0x20?

ast# boot
aspeednic#0: PHY at 0x20
set_mac_control_register 1453
Retry: Command = 2, Response_Code = 0
Retry: Command = 2, Response_Code = 0
Retry: Command = 1, Response_Code = 0
Retry: Command = 1, Response_Code = 0
Retry: Command = 2, Response_Code = 0
Retry: Command = 2, Response_Code = 0
Retry: Command = 1, Response_Code = 0

In uboot, I dumped the RxRing and found all of them are filled by pause frame. Something that the Broadcom adpater tries to tell BMC: I'm running out of free buffer, don't send me any more packets.

ast# md.l 5efec300 4
5efec300: b1010040 00000001 00000000 5efea9e0    @..............^
ast# md.l 5efea9e0 20
5efea9e0: 00c28001 80010100 010000c2 01000888    ................
5efea9f0: 0000ffff 00000000 00000000 00000000    ................
5efeaa00: 00000000 00000000 00000000 00000000    ................
5efeaa10: 00000000 00000000 00000000 d6d70786    ................
5efeaa20: 00000000 00000000 00000000 00000000    ................
5efeaa30: 00000000 00000000 00000000 00000000    ................
5efeaa40: 00000000 00000000 00000000 00000000    ................
5efeaa50: 00000000 00000000 00000000 00000000    ................
[uboot_pkt_dump.txt](https://github.com/openbmc/openbmc/files/798129/uboot_pkt_dump.txt)

gwshan, 2017-03-09:
One Broadcom support was identified with help from @benh. One email was sent to him to check why the BCM5718 keeps sending PAUSE frame. I'll update when getting any responses.

karthikbgt, 2017-04-05:
Hi,
Is this ftgmac issue still open for AST2520 based OpenBMC.( Linux bmc 4.1.15 #1 Fri Oct 14 22:08:32 PDT 2016 armv6l GNU/Linux )

I get the attached error when i try to an SCP of a file via ethernet interface to OpenBMC running in AST2520 from an external server. So far, i have faced this issue whenever i try to do an SCP.

Can you kindly provide me some pointers on what should be taken care and steps to overcome this issue.

Kindly find the error logs .
[ErrorLog.txt](https://github.com/openbmc/openbmc/files/901036/ErrorLog.txt)

Thanks in advance,
Karthik

williamspatrick, 2017-04-06:
@karthikbgt - That 'uname' value is not a kernel we have ever used for OpenBMC. Where did that come from?

The ftgmac driver was recently updated and is going into our 4.10 tree (openbmc/Linux branch dev-4.10). We expect this driver to be much faster and it also supports the pause frames. This should be landing in upstream 4.12 if I understand correctly.

karthikbgt, 2017-04-06:
@williamspatrick - It is present in the box that is with me. Does it mean that it is not part of main stream kernel and i need to migrate to the newer version of kernel ? Kindly confirm.

williamspatrick, 2017-04-08:
Sorry, but I don't think you are actually running. OpenBMC. We only ever supported 4.4 and 4.7. The 25xx chips were only supported on 4.7. 

What system is this?  Can you give the contents of /etc/os-release?

karthikbgt, 2017-04-11:
@williamspatrick Thanks for your inputs. The file /etc/os-release is not existing in the system that is with me. However, the /etc/issue seems to convey as 'OpenBMC Release cmm-v11' ... Kindly clarify whether this is a different version.