openbmc/openbmc #187: bootcount is not changing for failures in Hostboot

causten, 2016-02-20:
147 is a system with a lot of problems.  Seems like there is no working memory dimms.  It turns out that it constantly restarts the ipl.  Boot Count is suppose to determine that we are only trying 3 times before we but the system in a failed state.  Because of the lack of stopping the system event log gets filled with 7xreboot.  A system trying to reboot over night can easily create over 400 event logs.

anoo1, 2016-03-01:
How do you recreate, is it a continuous power on/off? Want to verify this pull request addresses it:
https://github.com/openbmc/phosphor-host-ipmid/pull/70

anoo1, 2016-03-01:
I just remembered, it's with the pnor image that doesn't support the old dimms. Will try it out.

anoo1, 2016-05-06:
Boot count fixed with https://github.com/openbmc/phosphor-host-ipmid/pull/70
The system would still try to power back on after the watchdog expires (since hostboot fails to start up, a stop watchdog request is not sent). This would be a different discussion to see if hostboot expects the BMC to stop the watchdog if they fail to start.

closed 2016-05-16 by anoo1