openbmc/openbmc #190: Segfault encountered when rebooting

KennethWilke, 2016-02-25:
Hello!

I ran into a segfault today when rebooting a BMC, I'm not 100% sure what version this box was on but I believe it was a build from the master branch within the last couple of days. I will build a new image based on 927d81657e58292f9c1226e76c85f3a23a409b6b and see if I run into this again, but wanted to report just in case.

```
root@barreleye:/tmp# reboot
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Network.
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Stopped target Timers.
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped target Swap.
         Stopping Network Time Synchronization...
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[  OK  ] Stopped target Multi-User System.
         Stopping Login Service...
         Stopping Kernel Logging Service...
         Stopping Phosphor OpenBMC DBus REST daemon...
         Stopping System Logging Service...
         Stopping Settings DBUS Object...
         Stopping Phosphor OpenBMC user management daemon..         Stopping Serial Getty on ttyS4...
         Stopping Phosphor OpenBMC BT to DBUS...
         Stopping Network Service...
         Stopping Network DBUS object...
         Stopping Phosphor OpenBMC IPMI daemon...
         Stopping DBUS introspecting REST server....
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Phosphor OpenBMC event management daemon.
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped Phosphor OpenBMC IPMI daemon.
[  OK  ] Stopped Phosphor OpenBMC user management daemon.
[  OK  ] Stopped DBUS introspecting REST server..
[  OK  ] Stopped Settings DBUS Object.
[  OK  ] Stopped Network DBUS object.
[  OK  ] Stopped Phosphor OpenBMC BT to DBUS.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Serial Getty on ttyS4.
[  OK  ] Stopped Phosphor OpenBMC DBus REST daemon.
         Stopping Phosphor OpenBMC DBus service management daemon...
[  OK  ] Removed slice system-serial\x2dgetty.slice.
         Stopping Temp placeholder for skeleton function...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /run/initramfs/rw...
         Unmounting /run/initramfs/ro...
[  OK  ] Stopped Phosphor OpenBMC DBus service management daemon.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped Temp placeholder for skeleton function.
[  OK  ] Unmounted /run/initramfs/rw.
[  OK  ] Unmounted /run/initramfs/ro.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Stopped Dispatch Password Requests to Console Directory Watch.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Closed Syslog Socket.
[  OK  ] Reached target Shutdown.
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[528]: Received SIGTERM from PID 1 (systemd-shutdow).
BUG: Bad page map in process systemd-journal  pte:40b58100 pmd:4e42c831
[... 79 lines cut ...]
[<c005ecf0>] (delete_from_page_cache) from [<c0069428>] (truncate_inode_page+0x98/0xa4)
[<c0069428>] (truncate_inode_page) from [<c00695cc>] (truncate_inode_pages_range+0x168/0x518)
[<c00695cc>] (truncate_inode_pages_range) from [<c0069990>] (truncate_inode_pages+0x14/0x1c)
[<c0069990>] (truncate_inode_pages) from [<c00a1038>] (evict+0x90/0x128)
[<c00a1038>] (evict) from [<c00a1100>] (dispose_list+0x30/0x3c)
[<c00a1100>] (dispose_list) from [<c00a1d3c>] (evict_inodes+0xb4/0xbc)
[<c00a1d3c>] (evict_inodes) from [<c008e5b0>] (generic_shutdown_super+0x40/0xc0)
[<c008e5b0>] (generic_shutdown_super) from [<c008e810>] (kill_block_super+0x18/0x68)
[<c008e810>] (kill_block_super) from [<c008e910>] (deactivate_locked_super+0x44/0x74)
[<c008e910>] (deactivate_locked_super) from [<c00a42ac>] (cleanup_mnt+0x4c/0x70)
[<c00a42ac>] (cleanup_mnt) from [<c002a97c>] (task_work_run+0x6c/0x80)
[<c002a97c>] (task_work_run) from [<c000cba8>] (do_work_pending+0xa4/0xc0)
[<c000cba8>] (do_work_pending) from [<c000a2d0>] (slow_work_pending+0xc/0x20)
Code: e121f002 e594300c e3530000 ba000000 (e7f001f2)
---[ end trace c55bffe6d850907a ]---
Segmentation fault
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=126380k,nr_inodes=31595,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
```

nkskjames, 2016-02-26:
I believe those seg faults will go away now that we have gone to jffs2 filesystem.

nkskjames, 2016-02-26:
Scratch that.   I just hit a seg fault with jffs2.  I'll turn up the priority on investigating this issue.

shenki, 2016-02-29:
@nkskjames just some ideas:

I was poking around our uboot code, and there is a 'dramtest' command. It requires an argument, but doesn't do anything with it. Perhaps give this a go on one of your failing barreleyes:

```
boot# dramtest 1
```

https://github.com/openbmc/u-boot/blob/v2013.07-aspeed-openbmc/arch/arm/cpu/arm926ejs/aspeed/STRESS.c#L106

It's using the memory test functionality of the dram controller. I ran it on a palmetto here and it didn't fail.

Also, what is the status of these machines with respect to the uboot version? Do they contain the UART fix, or the SOC vs CPU watchdog reset changes:

 https://github.com/openbmc/u-boot/commit/4b44678c11ac82c7c797bb115e276181752ad54d

nkskjames, 2016-03-03:
On Failing EVT system with this commit: 927d81657e58292f9c1226e76c85f3a23a409b6b

boot# dramtest 1
**************************************************** 
*** ASPEED Stress DRAM                           *** 
***                          20131107 for u-boot *** 
**************************************************** 

Pattern = 8F5C57B5 : Pass 1/1
boot# dramtest 1
**************************************************** 
*** ASPEED Stress DRAM                           *** 
***                          20131107 for u-boot *** 
**************************************************** 

Pattern = C7868A5D : Pass 1/1
boot# dramtest 1
**************************************************** 
*** ASPEED Stress DRAM                           *** 
***                          20131107 for u-boot *** 
**************************************************** 

Pattern = 2D45E0C9 : Pass 1/1
boot#

KennethWilke, 2016-03-09:
Hit this bug again on today's master https://github.com/openbmc/openbmc/commit/d7eea1d1c79e85538ab775dc7fc25b44d26c82c0

I think there was some suspicion that this bug may be triggered/caused by network activity (or that may have been a different bug). In this case this BMC was wired directly to an ethernet port on a Barreleye server with no traffic flowing over it.

This reboot wasn't very long after powering on the BMC, I think all I did between powering on and off was cat the `hconsole` script to verify the reconnect code was part of this revision.

Here's todays console dump in case it helps:
```
root@barreleye:~# reboot
         Stopping Network Time Synchronization...
[  OK  ] Stopped target Timers.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target Network.
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Sockets.
         Stopping Load/Save Random Seed...
[  OK  ] Closed dropbear.socket.
[  OK  ] Stopped target Swap.
         Unmounting /run/initramfs/ro...
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Removed slice system-getty.slice.
         Unmounting /run/initramfs/rw...
[  OK  ]          Stopping Serial Getty on ttyS4...
         Stopping Network DBUS object...
         Stopping Phosphor OpenBMC user management daemon...
         Stopping Network Service...
BUG: Bad page map in process klogd  pte:40b42100 pmd:4ddc6831
addr:4e400000 vm_flags:00000075 anon_vma:  (null) mapping:cf472974 index:80
file:libc-2.22.so fault:filemap_fault mmap:generic_file_readonly_mmap readpage:squashfs_readpage
CPU: 0 PID: 931 Comm: klogd Not tainted 4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
[<c000f28c>] (unwind_backtrace) from [<c000cf0c>] (show_stack+0x10/0x14)
[<c000cf0c>] (show_stack) from [<c00787c8>] (print_bad_pte+0x158/0x190)
[<c00787c8>] (print_bad_pte) from [<c0079cec>] (unmap_single_vma+0x4ec/0x500)
[<c0079cec>] (unmap_single_vma) from [<c007a438>] (unmap_vmas+0x44/0x54)
[<c007a438>] (unmap_vmas) from [<c007e97c>] (exit_mmap+0xc8/0x1dc)
[<c007e97c>] (exit_mmap) from [<c0014770>] (mmput+0x38/0xb8)
[<c0014770>] (mmput) from [<c0017380>] (do_exit+0x2f0/0x7b0)
[<c0017380>] (do_exit) from [<c0018654>] (do_group_exit+0x4c/0xb8)
[<c0018654>] (do_group_exit) from [<c00205c8>] (get_signal+0x440/0x4a0)
[<c00205c8>] (get_signal) from [<c000c6e0>] (do_signal+0x8c/0x3a4)
[<c000c6e0>] (do_signal) from [<c000cb60>] (do_work_pending+0x5c/0xc0)
[<c000cb60>] (do_work_pending) from [<c000a2d0>] (slow_work_pending+0xc/0x20)
Disabling lock debugging due to kernel taint
         Stopping Kernel Logging Service...
         Stopping DBUS introspecting REST server....
         Stopping Phosphor OpenBMC IPMI daemon...
         Stopping System Logging Service...
         Stopping Phosphor OpenBMC DBus REST daemon...
         Stopping Settings DBUS Object...
         Stopping Phosphor OpenBMC BT to DBUS...
         Stopping Login Service...
         Stopping Phosphor OpenBMC event management daemon...
         Stopping Phosphor Inverse ARP daemon...
[  OK  BUG: Bad rss-counter state mm:ce463920 idx:0 val:1
] Stopped NetworBUG: Bad rss-counter state mm:ce463920 idx:2 val:-1
k Time Synchronization.
[  OK  ] Stopped Phosphor Inverse ARP daemon.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped DBUS introspecting REST server..
[  OK  ] Stopped Phosphor OpenBMC user management daemon.
[  OK  ] Stopped Serial Getty on ttyS4.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Phosphor OpenBMC BT to DBUS.
[  OK  ] Stopped Settings DBUS Object.
[  OK  ] Stopped Phosphor OpenBMC event management daemon.
[  OK  ] Stopped Network DBUS object.
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Phosphor OpenBMC IPMI daemon.
[... 86 lines cut ...]
[<c005ecfc>] (delete_from_page_cache) from [<c0069430>] (truncate_inode_page+0x98/0xa4)
[<c0069430>] (truncate_inode_page) from [<c00695d4>] (truncate_inode_pages_range+0x168/0x518)
[<c00695d4>] (truncate_inode_pages_range) from [<c0069998>] (truncate_inode_pages+0x14/0x1c)
[<c0069998>] (truncate_inode_pages) from [<c00a1060>] (evict+0x90/0x128)
[<c00a1060>] (evict) from [<c00a1128>] (dispose_list+0x30/0x3c)
[<c00a1128>] (dispose_list) from [<c00a1d64>] (evict_inodes+0xb4/0xbc)
[<c00a1d64>] (evict_inodes) from [<c008e5d0>] (generic_shutdown_super+0x40/0xc0)
[<c008e5d0>] (generic_shutdown_super) from [<c008e830>] (kill_block_super+0x18/0x68)
[<c008e830>] (kill_block_super) from [<c008e930>] (deactivate_locked_super+0x44/0x74)
[<c008e930>] (deactivate_locked_super) from [<c00a42d4>] (cleanup_mnt+0x4c/0x70)
[<c00a42d4>] (cleanup_mnt) from [<c002a97c>] (task_work_run+0x6c/0x80)
[<c002a97c>] (task_work_run) from [<c000cba8>] (do_work_pending+0xa4/0xc0)
[<c000cba8>] (do_work_pending) from [<c000a2d0>] (slow_work_pending+0xc/0x20)
Code: e121f002 e594300c e3530000 ba000000 (e7f001f2)
---[ end trace 38d01c1982ab9e36 ]---
Segmentation fault
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=126364k,nr_inodes=31591,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
```

mdmillerii, 2016-04-01:
The belief is these should be fixed by the u-boot stop-network patches.  The most recent report stated the network was connected to a switch so it could have gotten a link up/down report or a packet from the barreleye host when it was started.

shenki, 2016-06-06:
We think this is resolved with the u-boot fix that is now in the tree. Please reopen if you see the symptoms again.

williamspatrick, 2016-06-14:
We believe this is fixed.  Please reopen if not.

closed 2016-06-14 by williamspatrick