openbmc/openbmc #191: OCC errors on host shutoff

KennethWilke, 2016-02-25:
Howdy OpenBMC folks,

We've noticed that sometimes we'll get a bunch of OCC related errors after a shutoff, and wanted to see if this is expected behavior.

```
root@barreleye:~# obmcutil poweroff
occ-i2c 3-0051: ERROR: sensor type FREQ not supported
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -1
 = None
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
root@barreleye:~# occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
occ-i2c 3-0051: ERROR: OCC data length is zero
occ-i2c 3-0051: ERROR: cannot get occ sensor data: -22
```

nkskjames, 2016-02-25:
This is known behavior.  Basically we disable the OCC driver after we get a host powered off message.  There is a window between when host is powered off and this disabling of the driver when it still tries to read the OCC, but it can't because host is powered off.  It won't hurt anything and one day it will be cleaned up.

KennethWilke, 2016-02-25:
@nkskjames mike says hi

should we close this issue since it's known behavior, or would you like to leave it open as a reminder to clean up?

nkskjames, 2016-02-26:
Leave it open until we fix the errors.

adamliyi, 2016-03-02:
@nkskjames , ok. I will clean up these message.

adamliyi, 2016-03-03:
Fixed by PR: https://github.com/openbmc/linux/pull/56

closed 2016-03-03 by nkskjames