openbmc/openbmc #194: inventory_items error

KennethWilke, 2016-03-01:
This morning I build a new OpenBMC based on 927d81657e58292f9c1226e76c85f3a23a409b6b and saw this message on my first boot, this was while I using an `obmcutil` command, but the box was still starting it's services. Attached is the error I received and my process list which shows the `inventory_items` program and a few others in a zombie state

```
root@barreleye:~# obmcutil state
------------[ cut here ]------------
Kernel BUG at c00986b8 [verbose debug info unavailable]
Internal error: Oops - BUG: 0 [#1] ARM
Modules linked in:
CPU: 0 PID: 970 Comm: inventory_items Not tainted 4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
task: cf9cd3e0 ti: cdef6000 task.ti: cdef6000
PC is at putname+0x10/0x64
LR is at do_sys_open+0x19c/0x1c0
pc : [<c00986b8>]    lr : [<c008c644>]    psr: 60000013
sp : cdef7f68  ip : 00000009  fp : b6bd67b0
r10: 00000000  r9 : cf75e1c0  r8 : cdd94f28
r7 : ce4df000  r6 : 00000020  r5 : cdd94f20  r4 : 00000009
r3 : 00000000  r2 : 00000000  r1 : ce4df000  r0 : ce4df000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 4df34000  DAC: 00000051
Process inventory_items (pid: 970, stack limit = 0xcdef6190)
Stack: (0xcdef7f68 to 0xcdef8000)
7f60:                   00000000 00000000 00000013 00020241 000081b6 00000022
7f80: 00000300 00000001 cf8030c0 00104b88 00000004 00000004 00000005 c000a424
7fa0: cdef6000 c000a280 00104b88 00000004 00126068 00020241 000001b6 000001b6
7fc0: 00104b88 00000004 00000004 00000005 4310d4c0 00126068 0011d5d8 b6bd67b0
7fe0: 00000000 be82e074 42d6c198 42dc495c 60000010 00126068 00000000 00000000
[<c00986b8>] (putname) from [<00000000>] (  (null))
Code: e590300c e1a01000 e3530000 ca000000 (e7f001f2)
---[ end trace 33b136da35e22d9d ]---

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:08 UTC):

kernel[920]: Internal error: Oops - BUG: 0 [#1] ARM

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: Process inventory_items (pid: 970, stack limit = 0xcdef6190)

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: Stack: (0xcdef7f68 to 0xcdef8000)

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: 7f60:                   00000000 00000000 00000013 00020241 000081b6 00000022

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: 7f80: 00000300 00000001 cf8030c0 00104b88 00000004 00000004 00000005 c000a424

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: 7fa0: cdef6000 c000a280 00104b88 00000004 00126068 00020241 000001b6 000001b6

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: 7fc0: 00104b88 00000004 00000004 00000005 4310d4c0 00126068 0011d5d8 b6bd67b0

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: 7fe0: 00000000 be82e074 42d6c198 42dc495c 60000010 00126068 00000000 00000000

Broadcast message from systemd-journald@barreleye (Thu 2016-02-25 15:17:09 UTC):

kernel[920]: Code: e590300c e1a01000 e3530000 ca000000 (e7f001f2)

 = BMC_STARTING2
root@barreleye:~# ps
  PID USER       VSZ STAT COMMAND
    1 root      5044 S    {systemd} /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [kdevtmpfs]
    8 root         0 SW<  [writeback]
    9 root         0 SW   [kworker/u2:1]
   87 root         0 SW   [kworker/u2:2]
  129 root         0 SW   [kworker/u2:3]
  142 root         0 SW<  [crypto]
  144 root         0 SW<  [bioset]
  146 root         0 SW<  [kblockd]
  185 root         0 SW<  [rpciod]
  186 root         0 SW   [kworker/0:1]
  193 root         0 SW   [kswapd0]
  194 root         0 SW   [fsnotify_mark]
  195 root         0 SW<  [nfsiod]
  248 root         0 SW<  [bioset]
  250 root         0 SW<  [bioset]
  251 root         0 SW<  [bioset]
  252 root         0 SW<  [bioset]
  262 root         0 SW<  [bioset]
  265 root         0 SW<  [bioset]
  268 root         0 SW<  [bioset]
  271 root         0 SW<  [bioset]
  274 root         0 SW<  [bioset]
  277 root         0 SW<  [bioset]
  280 root         0 SW<  [bioset]
  283 root         0 SW<  [bioset]
  297 root         0 SW<  [bioset]
  302 root         0 SW<  [bioset]
  307 root         0 SW<  [bioset]
  312 root         0 SW<  [bioset]
  317 root         0 SW<  [bioset]
  322 root         0 SW<  [bioset]
  327 root         0 SW<  [bioset]
  333 root         0 SW<  [bioset]
  458 root         0 SW<  [deferwq]
  487 root         0 RWN  [jffs2_gcd_mtd6]
  492 root         0 SW<  [kworker/0:1H]
  493 root         0 SW   [kworker/0:2]
  509 root      6708 S    /lib/systemd/systemd-journald
  513 root         0 SW   [kworker/0:3]
  522 root         0 SW   [kworker/0:4]
  524 root         0 SW   [kworker/0:5]
  529 root         0 SW   [kworker/0:6]
  847 root     10572 S    /lib/systemd/systemd-udevd
  897 systemd- 13036 S    /lib/systemd/systemd-timesyncd
  918 root     15144 S    python /usr/sbin/phosphor-mapper
  920 root      2968 S    /sbin/klogd -n
  921 root      3324 S    -sh
  922 root     16488 S    {system_manager.} /usr/bin/python -u /usr/sbin/syste
  923 root      3032 S    /sbin/syslogd -n -C
  924 root      4940 S    /usr/sbin/ipmid
  926 root     14804 S    {settings_manage} /usr/bin/python -u /usr/sbin/setti
  927 root     12964 S    python /usr/sbin/rest-dbus
  928 root     15080 S    python /usr/sbin/obmc-phosphor-userd
  929 root     14904 S    python /usr/sbin/netman.py
  930 root      4536 S    /usr/sbin/obmc-phosphor-eventd
  931 root      3636 S    /usr/sbin/btbridged
  932 messageb  3320 S    /usr/bin/dbus-daemon --system --address=systemd: --n
  935 systemd-  3480 S    /lib/systemd/systemd-networkd
  936 root      2884 S    /lib/systemd/systemd-logind
  937 root     15088 S    {sensor_manager2} /usr/bin/python -u /usr/sbin/senso
  943 root     43340 S    python /usr/sbin/obmc-rest
  946 root     24248 S    /usr/sbin/power_control.exe 3000 10
  948 root     24244 S    /usr/sbin/control_host.exe
  951 root     14844 S    {hwmon.py} /usr/bin/python -u /usr/sbin/hwmon.py Bar
  952 root      3632 S    /usr/sbin/led_controller.exe
  953 root     14880 S    {download_manage} /usr/bin/python -u /usr/sbin/downl
  954 root     14664 S    {bmc_update.py} /usr/bin/python -u /usr/sbin/bmc_upd
  955 root     24244 S    /usr/sbin/button_reset.exe
  956 root     24244 S    /usr/sbin/button_power.exe
  957 root     24236 S    /usr/sbin/host_watchdog.exe
  958 root     24236 S    /usr/sbin/control_bmc.exe
  961 root     24244 S    /usr/sbin/flash_bios.exe
  970 root         0 Z    [inventory_items]
 1006 root     14568 S    {chassis_control} /usr/bin/python -u /usr/sbin/chass
 1007 root     14480 S    {fan_control.py} /usr/bin/python -u /usr/sbin/fan_co
 1011 root         0 Z    [discover_system]
 1012 root         0 Z    [flasher.exe]
 1017 root         0 Z    [flasher.exe]
 1020 root      3440 R    ps
root@barreleye:~#
```

williamspatrick, 2016-06-15:
This is a kernel oops.

mdmillerii, 2016-06-15:
Similar errors were observed among the several symptoms found before the i2c driver fix was merged.  I haven't heard of it since but don't know how to directly relate it to that bug.

shenki, 2016-06-15:
This is an old kernel release. Please close

closed 2016-06-15 by williamspatrick