openbmc/openbmc #196: Power Restoration Policy is not working as expected

ratagupt, 2016-03-03:
1) System was Powered On.. 
2) Power_policy was set to LEAVE_OFF 
3) I did a PDU reboot 
4) when it came back, and checked the host settings system state said it was POWERED_ON

vishwabmc, 2016-05-30:
I am looking into it.. The discover_system_state.py has checks for below only and does not mention anything like LEAVE_OFF.

1) ALWAYS_POWER_ON
2) RESTORE_LAST_STATE
3) HOST_POWERED_ON

Need to see what are the correct enums for power restore policy.

vishwabmc, 2016-06-27:
I executed this test on Barreleye ( Barreleye PVT system ) with 4.4.12-openbmc-20160606-1 and I did not see it getting recreated.. I see that PGOOD reads OFF and hence results in staying in BMC_READY.

This may be a Palmetto only issue ??

vishwabmc, 2016-06-27:
And.. I tried it on palm205 just now that is running 4.4.12-openbmc-20160606-1 and I don't see the issue reported.. 

System was HOST_BOOTING and I had set the power_policy as LEAVE_OFF and the system stayed in OFF.
root@palmetto:~# python /usr/sbin/discover_system_state.py 
VISHWA PGOOD IS OFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
Last System State: HOST_POWERED_OFF;  Power Policy: LEAVE_OF

vishwabmc, 2016-06-27:
root@barreleye:~# journalctl --no-pager | grep LEAVE_OFF
Jun 07 00:53:06 barreleye system_manager.py[987]: Last System State: ;  Power Policy: LEAVE_OFF
root@barreleye:~# obmcutil state
 = BMC_READY

that is on Barreleye.. 

So with these exercises, I can concluded we do not have the issue.. May be the issue is fixed by some code in these days / the issue reported may have been one off ??

vishwabmc, 2016-06-27:
I also noticed that discover_system_state.py is not being called in Palmetto as I see only these :+1: 

$ grep -r discover_system_state *
configs/Garrison.py:        'process_name'    : 'discover_system_state.py',
configs/Firestone.py:        'process_name'    : 'discover_system_state.py',
configs/Witherspoon.py:        'process_name'    : 'discover_system_state.py',
configs/Barreleye.py:           'process_name'    : 'discover_system_state.py',
pystatemgr/setup.py:      scripts=['discover_system_state.py', 'goto_system_state.py'],

williamspatrick, 2016-06-27:
"4.4.12-openbmc-20160606-1"  -- what does this mean?

@bradbishop - Any idea why Palmetto is missing the 'discover_system_state.py' call?

vishwabmc, 2016-06-27:
@williamspatrick  "4.4.12-openbmc-20160606-1 is the output from uname -a

williamspatrick, 2016-06-27:
@vishwabmc - Oh.  That is a kernel level and not a openbmc code level.  See /etc/os-release for an openbmc code level description.

vishwabmc, 2016-06-29:
The conclusion is : Without the discover_system_state.py getting invoked, I am not sure how it can do any state change beyond BMC_READY.  So the issue reported here may be a one off since it was opened sometime before and the skeleton has taken many changes since then.

I would vote for opening a new issue to add the missing discover_system_state.py into Palmetto.py since I see that change is doing the needed transitions under different host states and power policies.

williamspatrick, 2016-06-29:
Two steps here:

1. Vishwa to make change to enable discover_system_state.py on Palmetto.
2. Re-test all the different states.  Vishwa suspects RESTORE_LAST_STATE doesn't work properly.

vishwabmc, 2016-06-30:
Since I am not able to recreate this, please move this to Ratan so he could put his comments and close.

If RESTORE_LAST_STATE is not working fine, I will open a new issue just for that.

williamspatrick, 2016-08-16:
Code is in Gerrit.  Waiting for stable master to be able to test.  Held up on kernel issues.

williamspatrick, 2016-08-25:
@ratagupt - We have one fix in Gerrit that should allow us to get to BMC_READY state correctly.  Should be able to test your fix tomorrow.

closed 2016-09-27 by williamspatrick