openbmc/openbmc #202: Networking:- Ip address validation

ratagupt, 2016-03-07:
Some Ip addresses which are not valid so we should not add it on the interface
1) x.x.x.0
2) x.x.x.255
3)0.0.0.0

ratagupt, 2016-03-07:
The same verification should be valid for Gateway Parameter....

anoo1, 2016-05-23:
Doug can take a look.

williamspatrick, 2016-06-15:
This should handle ipv6 as well.

eddiejames, 2016-07-13:
Can anyone share which address ranges to filter for ipv6? Also are there some more to filter for ipv4 than just the ones stated? Thanks

mdmillerii, 2016-07-13:
Class D multicast should not be used for host or gateway addresses for ipv4, and probably class E.  See [rfc5735](https://tools.ietf.org/html/rfc5735).  We probably should not allow setting to the ipv6 relay range or this-network range either, and we require a helper to do link local addresses.

Note that the above all-zero and all-ones adresses should be based on the subnet mask.

closed 2016-08-02 by williamspatrick