openbmc/openbmc #206: Zombie processes on startup

KennethWilke, 2016-03-09:
Howdy!

Today I'm doing some OpenBMC testing on Barreleye using the current master https://github.com/openbmc/openbmc/commit/d7eea1d1c79e85538ab775dc7fc25b44d26c82c0

I'm seeing a few zombie processes in the process list shortly after startup, and wanted to report it as that seems unusual.

My full process list below:
```
root@barreleye:~# ps
  PID USER       VSZ STAT COMMAND
    1 root     23420 S    {systemd} /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [kdevtmpfs]
    8 root         0 SW<  [writeback]
    9 root         0 SW   [kworker/u2:1]
  142 root         0 SW<  [crypto]
  144 root         0 SW<  [bioset]
  146 root         0 SW<  [kblockd]
  183 root         0 SW<  [rpciod]
  184 root         0 SW   [kworker/0:1]
  193 root         0 SW   [kswapd0]
  194 root         0 SW   [fsnotify_mark]
  195 root         0 SW<  [nfsiod]
  249 root         0 SW<  [bioset]
  250 root         0 SW<  [bioset]
  251 root         0 SW<  [bioset]
  252 root         0 SW<  [bioset]
  262 root         0 SW<  [bioset]
  265 root         0 SW<  [bioset]
  268 root         0 SW<  [bioset]
  271 root         0 SW<  [bioset]
  274 root         0 SW<  [bioset]
  277 root         0 SW<  [bioset]
  280 root         0 SW<  [bioset]
  283 root         0 SW<  [bioset]
  297 root         0 SW<  [bioset]
  302 root         0 SW<  [bioset]
  307 root         0 SW<  [bioset]
  312 root         0 SW<  [bioset]
  317 root         0 SW<  [bioset]
  322 root         0 SW<  [bioset]
  327 root         0 SW<  [bioset]
  333 root         0 SW<  [bioset]
  459 root         0 SW<  [deferwq]
  505 root         0 SWN  [jffs2_gcd_mtd6]
  510 root         0 SW<  [kworker/0:1H]
  515 root         0 SW   [kworker/0:2]
  532 root      6708 S    /lib/systemd/systemd-journald
  867 root     10440 S    /lib/systemd/systemd-udevd
  874 systemd- 13036 S    /lib/systemd/systemd-timesyncd
  926 root      3636 S    /usr/sbin/btbridged
  927 root      3032 S    /sbin/syslogd -n -C
  929 root      1728 S    /usr/sbin/inarp
  930 root     16236 S    {system_manager.} /usr/bin/python -u /usr/sbin/system_manager.py Barreleye
  931 root     16992 S    python /usr/sbin/netman.py
  932 root     15360 S    python /usr/sbin/obmc-phosphor-userd
  933 messageb  3288 S    /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
  934 root      2884 S    /lib/systemd/systemd-logind
  935 root      2968 S    /sbin/klogd -n
  937 root      4536 S    /usr/sbin/obmc-phosphor-eventd
  938 root     14548 S    {settings_manage} /usr/bin/python -u /usr/sbin/settings_manager.py
  939 root     12964 S    python /usr/sbin/rest-dbus
  940 root     15528 S    python /usr/sbin/phosphor-mapper
  941 systemd- 11672 S    /lib/systemd/systemd-networkd
  943 root      4944 S    /usr/sbin/ipmid
  953 root     15088 S    {sensor_manager2} /usr/bin/python -u /usr/sbin/sensor_manager2.py Barreleye
  958 root     43340 S    python /usr/sbin/obmc-rest
  961 root     24248 S    /usr/sbin/power_control.exe 3000 10
  963 root     24244 S    /usr/sbin/control_host.exe
  965 root     14844 S    {hwmon.py} /usr/bin/python -u /usr/sbin/hwmon.py Barreleye
  966 root      3628 S    /usr/sbin/led_controller.exe
  967 root     14880 S    {download_manage} /usr/bin/python -u /usr/sbin/download_manager.py Barreleye
  968 root     14664 S    {bmc_update.py} /usr/bin/python -u /usr/sbin/bmc_update.py
  969 root     24244 S    /usr/sbin/button_reset.exe
  970 root     24244 S    /usr/sbin/button_power.exe
  971 root     24236 S    /usr/sbin/host_watchdog.exe
  972 root     24236 S    /usr/sbin/control_bmc.exe
  973 root     24244 S    /usr/sbin/flash_bios.exe
  974 root     15040 S    {inventory_items} /usr/bin/python -u /usr/sbin/inventory_items.py Barreleye
 1019 root     14568 S    {chassis_control} /usr/bin/python -u /usr/sbin/chassis_control.py
 1020 root     14480 S    {fan_control.py} /usr/bin/python -u /usr/sbin/fan_control.py
 1024 root         0 Z    [discover_system]
 1025 root         0 Z    [flasher.exe]
 1030 root         0 Z    [flasher.exe]
 1042 root      2880 R    /usr/sbin/dropbear -i -r /etc/dropbear/dropbear_rsa_host_key -B
 1043 root      3324 S    -sh
 1065 root         0 SW   [kworker/0:0]
 1067 root         0 SW   [kworker/0:3]
 1068 root      2228 S    /sbin/agetty -8 -L --keep-baud ttyS4 115200 xterm
 1072 root      3440 R    ps
```

jk-ozlabs, 2016-03-09:
@KennethWilke thanks for posting; we should take a look at these. Even if they're harmless, it makes it difficult to tell if something else has actually gone wrong.

KennethWilke, 2016-05-16:
As a follow up on this one, on the `0.6.1` tag:

 `flasher-exe` processes are both child processes of `/usr/sbin/flash_bios.exe`

There is also now a zombie process `pcie_slot_prese` that is a child process of `{system_manager.} /usr/bin/python -u /usr/sbin/system_manager.py Barreleye`

adamliyi, 2016-08-05:
flash_bios_obj.c already defines handler to handle SIGCHLD, and wait on the child process. Not sure why it was commented out. Submitted a patch to re-enable the handler. Tested on Barreleye, there is no 'flasher.exe' zombie processes.
```
root@barreleye:~# ps | grep -i 'z'
  PID USER       VSZ STAT COMMAND
 1072 root         0 Z    [discover_system]
 1095 root      2972 S    grep -i z
```

adamliyi, 2016-08-18:
Those zombie processes are child processes of "system_manager.py". SystemManager starts many child processes with 'popen()' when system state changes. Committed a patch to fix the zombie process issue: https://gerrit.openbmc-project.xyz/#/c/228/

Before the change:

```
root@barreleye:~# ps
  PID USER       VSZ STAT COMMAND
    1 root     13116 S    {systemd} /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [kdevtmpfs]
    8 root         0 SW<  [writeback]
   58 root         0 SW   [kworker/u2:2]
  141 root         0 SW<  [crypto]
  143 root         0 SW<  [bioset]
  145 root         0 SW<  [kblockd]
  184 root         0 SW<  [rpciod]
  192 root         0 SW   [kswapd0]
  193 root         0 SW   [fsnotify_mark]
  194 root         0 SW<  [nfsiod]
  248 root         0 SW<  [bioset]
  249 root         0 SW<  [bioset]
  250 root         0 SW<  [bioset]
  251 root         0 SW<  [bioset]
  261 root         0 SW<  [bioset]
  264 root         0 SW<  [bioset]
  267 root         0 SW<  [bioset]
  269 root         0 SW<  [bioset]
  272 root         0 SW<  [bioset]
  276 root         0 SW<  [bioset]
  279 root         0 SW<  [bioset]
  282 root         0 SW<  [bioset]
  296 root         0 SW<  [bioset]
  301 root         0 SW<  [bioset]
  306 root         0 SW<  [bioset]
  311 root         0 SW<  [bioset]
  316 root         0 SW<  [bioset]
  321 root         0 SW<  [bioset]
  326 root         0 SW<  [bioset]
  332 root         0 SW<  [bioset]
  468 root         0 SW<  [deferwq]
  514 root         0 SWN  [jffs2_gcd_mtd6]
  519 root         0 SW<  [kworker/0:1H]
  545 root      6704 R    /lib/systemd/systemd-journald
  880 root     10208 S    /lib/systemd/systemd-udevd
  886 systemd- 12932 S    /lib/systemd/systemd-timesyncd
  947 root     15244 S    python /usr/sbin/phosphor-mapper
  949 root      1736 S    /usr/sbin/obmc-console-server
  950 root      3636 S    /usr/sbin/btbridged
  951 root      2968 S    /sbin/klogd -n
  952 root     14840 S    python /usr/sbin/obmc-phosphor-userd
  953 root      3032 S    /sbin/syslogd -n -C
  955 root      2880 S    /lib/systemd/systemd-logind
  956 root      1724 S    /usr/sbin/inarp
  957 messageb  3136 S    /usr/bin/dbus-daemon --system --address=systemd: --n
  962 root      4544 S    /usr/sbin/obmc-phosphor-eventd -s 200000 -t 128
  967 root      4956 S    /usr/sbin/ipmid
  977 root     43660 S    python /usr/sbin/obmc-rest
 1071 root      3068 S    -sh
 1213 root         0 SW   [kworker/0:0]
 1257 root         0 SW   [kworker/0:1]
 1372 root         0 SW   [kworker/u2:3]
 1415 root         0 SW   [kworker/0:2]
 1418 root     15168 R    {system_manager.} /usr/bin/python -u /usr/sbin/syste
 1419 root     14244 S    {settings_manage} /usr/bin/python -u /usr/sbin/setti
 1420 root     16536 S    python /usr/sbin/netman.py
 1421 root      3636 S    /usr/sbin/netman_watch_dns /run/systemd/netif/state
 1425 root     14892 S    {sensor_manager2} /usr/bin/python -u /usr/sbin/senso
 1428 root     24228 S    /usr/sbin/button_power.exe
[... 81 lines cut ...]
  1514 root         0 SW   [kworker/u2:1]
  1536 root         0 SW   [kworker/0:0]
  1539 root     15168 S    {system_manager.} /usr/bin/python -u /usr/sbin/syste
  1540 root     14244 S    {settings_manage} /usr/bin/python -u /usr/sbin/setti
  1541 root     16536 S    python /usr/sbin/netman.py
  1542 root      3636 S    /usr/sbin/netman_watch_dns /run/systemd/netif/state
  1544 root     15004 S    {sensor_manager2} /usr/bin/python -u /usr/sbin/senso
  1547 root     24228 S    /usr/sbin/button_power.exe
  1549 root     24224 S    /usr/sbin/control_bmc.exe
  1551 root     24236 S    /usr/sbin/control_host.exe
  1552 root     14856 S    {hwmon.py} /usr/bin/python -u /usr/sbin/hwmon.py Bar
  1555 root     24232 S    /usr/sbin/host_checkstop.exe
  1558 root     24228 S    /usr/sbin/button_reset.exe
  1560 root     24232 S    /usr/sbin/flash_bios.exe
  1563 root     15056 S    {inventory_items} /usr/bin/python -u /usr/sbin/inven
  1565 root     24232 S    /usr/sbin/power_control.exe 3000 10
  1568 root      3632 S    /usr/sbin/led_controller.exe
  1570 root     14636 S    {download_manage} /usr/bin/python -u /usr/sbin/downl
  1573 root     16772 S    {bmc_update.py} /usr/bin/python -u /usr/sbin/bmc_upd
  1575 root     24228 S    /usr/sbin/host_watchdog.exe
  1583 root     14324 S    {chassis_control} /usr/bin/python -u /usr/sbin/chass
  1585 root     14236 S    {fan_control.py} /usr/bin/python -u /usr/sbin/fan_co
  1620 root         0 SW   [kworker/u2:2]
  1624 root      3208 R    ps
```

sivassrr, 2016-09-07:
VERSION_ID=v1.0.5-2-g7a3e2dc-dirty
Still see some zombie process (flasher.exe) at the time of READY and host booted state. 
And sync_inventory as zombie at the time of BMC READY state. 
root@barreleye:~# obmcutil state
 = BMC_READY
root@barreleye:~# ps | grep -i 'z'
  PID USER       VSZ STAT COMMAND
 1056 root         0 Z    [discover_system]
 1057 root         0 Z    [sync_inventory_]
 1058 root         0 Z    [sync_inventory_]
 1059 root         0 Z    [flasher.exe]
 1064 root         0 Z    [flasher.exe]
 1092 root      2972 S    grep -i z

root@barreleye:~# obmcutil state
 = HOST_BOOTED
root@barreleye:~#  ps | grep -i 'z'
  PID USER       VSZ STAT COMMAND
 1059 root         0 Z    [flasher.exe]
 1064 root         0 Z    [flasher.exe]
 1094 root         0 Z    [python]
 1101 root         0 Z    [pcie_slot_prese]
 1133 root      2972 S    grep -i z

williamspatrick, 2016-09-07:
You are testing stable. Fix went into master.

closed 2016-09-07 by williamspatrick