openbmc/openbmc #209: Ansible errors with OpenBMC

KennethWilke, 2016-03-17:
Howdy folks,

In the process of trying to use Ansible with OpenBMC, I ran across a few snags. It looks like the dropbear service doesn't support SFTP, but this can be mitigated with ansibles `scp_if_ssh` option. The next thing I ran into was an error related to some missing python modules:

```Traceback (most recent call last):
  File "/home/root/.ansible/tmp/ansible-tmp-1458244675.33-140278317885821/ping", line 253, in <module>
    from compiler import ast, parse
ImportError: No module named compiler```

This module normally ships along with the python interpreter, can this be added to the default build or is there a process I can use to add this to in-house builds?

Thanks!

mdmillerii, 2016-03-22:
As you found, **Ansible** can be configured tor `scp` instead of `sftp`.  Alternatively you can add `openssh-sftp-server` but I have not tested that.  The `python` package is split into several micro packages, if you require additonal components you will need to look at the recipes to find which ones you require.  Perhaps you can contribute back a `package-group` to enable whats needed for **Ansible**.

The following probably should be added to our docs repository maybe the cheat sheet:

# Customizing your image content

The recipes control what is included in your image.  An example is at `meta-phosphor/common/recipes-phosphor/images/obmc-phosphor-image-no-sysmgr.bb` modifies the image to remove something.  You add packages to the `IMAGE_FEATURES` variable or can request specific `PACKAGES` in `IMAGE_INSTALL`.  I find both `git grep` to look at file content and `git ls-tree -r HEAD | grep` to look for a file name useful. 

The **[BitBake User Manual](https://www.yoctoproject.org/docs/1.6/bitbake-user-manual/bitbake-user-manual.html)** is relatively small and contains the language overview, but I find it lacks on the actual variables and helpers for writing a recipe.  The **Yocto Project [MegaManual](http://www.yoctoproject.org/docs/1.8/mega-manual/mega-manual.html#new-recipe-writing-a-new-recipe)** has a section on writing a recipe; the `xml` source content is also in this tree.

bradbishop, 2016-03-24:
Hi Ken

The first step would be to create a recipe for Ansible.  Even if it doesn't actually do anything right now you can at least list the package dependencies here, and then when you include this recipe in the build the deps will get pulled in.

You can see how the python modules in the base python distro are split up looking at python-2.7-manifest.inc in yocto core.  For example the compiler module can be included like this from a recipe:

RDEPENDS_${PN} += 'python-compiler'

To get this new recipe included in a build, just drop this in your local.conf file:

IMAGE_INSTALL += ansible # assuming you named the recipe ansible.bb

In the longer term, you could look at having your own OE layer that appends this line to the obmc-phosphor-image recipe to avoid the local.conf step.  And/or we could do something with image features.  Or if it makes sense for everyone we could add a new image recipe to the phosphor OE layer that includes ansible.

-brad

williamspatrick, 2016-06-15:
We think this is outside the scope of the greater openbmc project.  You can add this as a custom recipe to get the extra dependencies added.

closed 2016-06-15 by williamspatrick