openbmc/openbmc #216: Eventlog service need to use conf file for setting dblog instead of hardcoding the size

mkumatag, 2016-03-29:
https://github.com/openbmc/openbmc/pull/214

IMO we should use EnvironmentFile entry under the service section and read the option from that file instead of hardcoding the value directly. This option will even help to change the value runtime and test it.

Reference - http://fedoraproject.org/wiki/Packaging:Systemd#EnvironmentFiles_and_support_for_.2Fetc.2Fsysconfig_files

causten, 2016-04-01:
The systemd config file has the value settable in there.  
meta-phosphor/common/recipes-phosphor/obmc-phosphor-event/files/obmc-phosphor-event.service

Are you thinking of a different technique?

mkumatag, 2016-04-02:
currently we have following code in meta-phosphor/common/recipes-phosphor/obmc-phosphor-event/files/obmc-phosphor-event.service for size

[Service]
ExecStart=/usr/sbin/obmc-phosphor-eventd -s 200000

Having value directly here will become very difficult to change it programmatically, instead it will good if we read same value from environment file like below:

[Service]
EnvironmentFile=-/etc/eventd.conf
ExecStart=/usr/sbin/obmc-phosphor-eventd $OPTIONS

And the file content for /etc/eventd.conf looks like below:

OPTIONS="-s 200000"

Following this approach we have 2 advantages - 

1. Easy to modify the value quickly and restart the service.
2. As we are not changing anything in the service files, so no need to daemon-reload for config change.

anoo1, 2016-05-17:
We need to add an additional argument for the max number of logs (added via https://github.com/openbmc/phosphor-event/pull/18), so this may be an opportunity to move the size and number into the config file.

anoo1, 2016-05-23:
Doug can take a look

causten, 2016-05-23:
This isn't a v1.0 Backlog request.  The service works the way it does now fine.  This is a feature change request not a bug.  Nothing wrong with the request but we can ship as it is.  I do believe there will be a review of the event log service for GA3 in which case the milestone can be changed.  Nothing to do here for v1.0 Backlog

causten, 2016-05-23:
I vote this moves to a different backlog.  Not a needed change for v1.0
 
 
Chris AustenPOWER Systems Enablement Manager(512) 286-5184 (T/L: 363-5184)

closed 2016-08-17 by williamspatrick