openbmc/openbmc #218: enumerate is now a keyword, method names should not match

causten, 2016-04-01:
Found this issue while using our obmc UI.  
https://github.com/LRRoberts0122/openbmc/issues/13

Once the /enumerate became part of the REST schema it can no longer be a dbus interface.  Need to remove org.openbmc.enumerate from /org/openbmc/inventory and anywhere else it is found.  The same information can be retrieved via /org/openbmc/inventory/enumerate

williamspatrick, 2016-04-01:
> LRRoberts0122/openbmc#13

The enumerate method was meant as a performance optimization for the enumerate REST interface. Can you give more of an explanation what the problem is?

Patrick

causten, 2016-04-01:
Brad mentioned the issue.  The existing REST interface doesn't handle it well
See the issue above for more details

anoo1, 2016-05-04:
@bradbishop , is this still an issue?

williamspatrick, 2016-06-22:
Need one update to the recipes and this can be closed.

closed 2016-06-22 by williamspatrick