openbmc/openbmc #221: Shrink python package.

williamspatrick, 2016-04-01:
@williamspatrick commented on [Fri Sep 18 2015](https://github.com/openbmc/meta-phosphor/issues/13)

11824	/usr/lib/python2.7
1504	/usr/lib/libpython2.7.so.1.0
676	/usr/lib/python2.7/lib-dynload/unicodedata.so

Do we need to install *.py or just *.pyc.  Can we do some build-time processing to ensure everything is pyc?

    root@qemuarm:/usr# find -name "*.pyc" | wc -l
    301
    root@qemuarm:/usr# find -name "*.py" | wc -l
    313

---

@shenki commented on [Mon Sep 21 2015](https://github.com/openbmc/meta-phosphor/issues/13#issuecomment-141889784)

Python 3?

---

@williamspatrick commented on [Mon Sep 21 2015](https://github.com/openbmc/meta-phosphor/issues/13#issuecomment-142038275)

@shenki - Can you elaborate? Is python 3 smaller?  We would need to ensure all the python applications and modules we want to use support python 3 before making that leap.

geissonator, 2017-05-23:
Need some discussion, this is 2.1 at least, maybe even 3.0.  How stretched are we on flash size?  Just removing the .py files should be fairly easy if that's what we want.

geissonator, 2018-08-20:
https://stackoverflow.com/questions/850630/python-py-pyo-pyc-which-can-be-eliminated-for-an-embedded-system looks relevant.  Just need .pyc and could probably remove some of the python libs we don't need.

geissonator, 2019-02-25:
We reduced the overall size of python by removing the majority of .py files. We're making progress with making python optional in the phosphor layer (https://github.com/ibm-openbmc/dev/issues/187). Closing this one out.

closed 2019-02-25 by geissonator