openbmc/openbmc #231: Host Console broken on Barreleye PVT Server

agangidi53, 2016-04-13:
Host console is broken on one of our Barreleye PVT server.
These are the commands I tried in the same order after booting the BMC up (racked the server in) :

   obmcutil state   
   obmcutil poweron
   hconsole

HCONSOLE OUTPUT:
Error getting authority: Error initializing authority: Could not connect: No such file or directory (g-io-error-quark, 1)

Welcome to emergency mode! After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or ^D to
try again to boot into default mode.

Cannot open access to console, the root account is locked.
See sulogin(8) man page for more details.

UNAME RESULT:
root@barreleye:~# uname -a
Linux barreleye 4.3.6-openbmc-20160222-1 #1 Thu Mar 24 09:14:41 CST 2016 armv5tejl GNU/Linux

OS RELEASE RESULT:
root@barreleye:~# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (master)
VERSION_ID=6.12
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (master)
BUILD_ID=v0.6-12-g19a1c40-dirty

agangidi53, 2016-04-13:
Attached are jorunalctl logs.[](url)
[debug_log_journalctl.txt](https://github.com/openbmc/openbmc/files/218072/debug_log_journalctl.txt)

nkskjames, 2016-04-13:
That is not a pure 0.6.1 build.  Something is very wrong.  Maybe start with a pure tagged level?

agangidi53, 2016-04-13:
Foxconn spun this out of pure 0.6.1 build I think. But this is the one that got deployed in DC.    May be Ken from Foxconn can comment on what else went in there.

Kenthliu, 2016-04-14:
@agangidi53  Does this issue happen in every board? We can not reproduce by our side. Does the system idle for a while? What's the timing you issue hconsole? Do you see BMC_READY in obmcutil state then obmcutil poweron? 
@nkskjames  The Build ID is because we use local build. This version is based on 19a1c40007d11cc0702a342702b66a77da523753 in v0.6-stable and merged some code like fan algorithm.

agangidi53, 2016-04-14:
@Kenthliu  No. This does NOT happen on every board but we have tried this build only on 2 PVT servers till now. This happens for first boot. Yes I wait for BMC_READY before doing a poweron. When I check after powering it on I get a HOST_BOOTED after sometime.

agangidi53, 2016-04-19:
Can we figure out why this happened from the logs ? I haven't been able to reproduce it on another server till now .

nkskjames, 2016-04-28:
@Kenthliu Is this still an issue?

agangidi53, 2016-04-28:
couldn't reproduce it ... let's close this. Will open it as a bug and reference this if I observe this again.

anoo1, 2016-05-17:
Closing per above comments.

closed 2016-05-17 by anoo1