openbmc/openbmc #234: version_id needs to be exposed via REST

causten, 2016-04-18:
Right now we are using the BUILD_ID as "version" property in /org/openbmc/inventory/system/chassis/motherboard/bmc

According to the systemd documentation BUILD_ID should not change between code updates.  
https://www.freedesktop.org/software/systemd/man/os-release.html#BUILD_ID=
So after a code update the use case we want is to show the new code version, not the old version. 

Adriana had a good use case of keeping the BUILD_ID around too.  That is to say it would be the golden side version in case of failures.  So I think in the REST/dbus interface the map should loook like this...

"version" == VERSION_ID
"build" == BUILD_ID

$ cat /etc/os-release 
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v0.6
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v0.6-19-gf363f61-dirty

anoo1, 2016-05-04:
Fixed with d8d9b4fdc1d781657200f0108dbb20f89d46c67c

anoo1, 2016-05-17:
Fixed with https://github.com/openbmc/skeleton/pull/63

closed 2016-05-17 by anoo1