openbmc/openbmc #235: TEMPLATECONF doesn't get picked up by oe-setup-builddir

GHF, 2016-04-20:
on my workstation, bitbake always builds for qemuarm regardless of what I set TEMPLATECONF to when sourcing oe-init-build-env. It's fixed upstream in OE but isn't part of the Yocto Jethro release.

http://git.openembedded.org/openembedded-core/commit/oe-init-build-env?id=0bf9d919bba5780d34cc6ac9dde2c74bb07b8342

http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/oe-init-build-env?h=jethro

williamspatrick, 2016-06-23:
This is picked up by Yocto 2.1, so it will be in our tree as soon as pull #396 is merged.
```
$ git log e8d36 -n1 | grep OE-Core
    (From OE-Core rev: 0bf9d919bba5780d34cc6ac9dde2c74bb07b8342)
$ git tag --contains e8d36f4886e88cc496b04890055fbc534c3caa6f
yocto-2.1
```

closed 2016-06-27 by williamspatrick

shenki, 2016-06-28:
Woot! Nice work Patrick. Hopefully keeping in line with Yocto releases will be easier from here on in.