openbmc/openbmc #236: squashfs corruption over time

shenki, 2016-04-21:
Adi mentioned they are seeing corruption after leaving the bmc to run for about 54 hours

```
SQUASHFS error: Unable to read page, block 59daf0, size adbc
free -m
SQUASHFS error: Unable to read data cache entry [61ee8]
SQUASHFS error: Unable to read page, block 61ee8, size f2b0
Bus error
root@barreleye:~# SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
df -SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
hSQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc

Filesystem                Size      Used Available Use% Mounted on
dev                     123.4M         0    123.4M   0% /dev
tmpfs                   124.3M     22.3M    102.0M  18% /run
/dev/mtdblock5           22.6M     22.6M         0 100% /run/initramfs/ro
/dev/mtdblock6            4.0M    440.0K      3.6M  11% /run/initramfs/rw
cow                       4.0M    440.0K      3.6M  11% /
tmpfs                   124.3M         0    124.3M   0% /dev/shm
tmpfs                   124.3M         0    124.3M   0% /sys/fs/cgroup
tmpfs                   124.3M    176.0K    124.1M   0% /tmp
tmpfs                   124.3M      8.0K    124.3M   0% /var/volatile
root@barreleye:~# SQUASHFS error: Unable to read fragment cache entry [59daf0]
SQUASHFS error: Unable to read page, block 59daf0, size adbc
SQUASHFS error: Unable to read fragment cache entry
```

mdmillerii, 2016-04-21:
From other reports the decompression fails causing the caches to be populated with error entries.  The file system in flash is ok when upon reboot.

We can try `dmesg` or a log capture at the start of the corruption, or add instrumentation of the data being decompressed.  Could be some error preventing the flash to be read or memory corruption maybe of the decompression workspace or cache pointing to data to be read.

shenki, 2016-04-21:
On 21 Apr 2016 11:22 pm, "Milton D. Miller II" <notifications@github.com>
wrote:
>
> From other reports the decompression fails causing the caches to be
populated with error entries. The file system in flash is ok when upon
reboot.

Where are the other reports? Can we collect them here please.

Have we seen it happen with development machines, or just those in the
datacentre?

>
> We can try dmesg or a log capture at the start of the corruption, or add
instrumentation of the data being decompressed. Could be some error
preventing the flash to be read or memory corruption maybe of the
decompression workspace or cache pointing to data to be read.

The sigbus in the log I pasted suggests memory corruption beyond bad flash
data.

mdmillerii, 2016-04-21:
I received a report Monday.  The claim was 12 systems were flashed Friday with PVT code and monitoring of the periodic (30s) ssh connect attempts was made and recorded.  

_Since we've had this kind of thing happen a few times I setup monitoring to keep an eye on the BMCs and it looks like 12 of them are down this morning, failing roughly in the order I reflashed them in (bottom of the rack, up)._

All failed over time with age reported 2 - 12 hours (guessing this was time of last good report).

The system with the debug serial cable was proded with this result:

'
SQUASHFS error: Unable to read metadata cache entry [16832a4]
SQUASHFS error: Unable to read inode 0x1648
barreleye login:
SQUASHFS error: Unable to read metadata cache entry [16832a4]
SQUASHFS error: Unable to read inode 0x1648
barreleye login:
SQUASHFS error: Unable to read metadata cache entry [16832a4]
SQUASHFS error: Unable to read inode 0x1648
barreleye login: rootSQUASHFS error: xz decompression failed, data probably corrupt
SQUASHFS error: squashfs_read_data failed to read block 0x3137f4
SQUASHFS error: Unable to read data cache entry [3137f4]
SQUASHFS error: Unable to read page, block 3137f4, size da98
SQUASHFS error: Unable to read metadata cache entry [1684b1a]
'

Note after root login the xz decompression report, which means the attempt to decompress a buffer failed.  The rest are reporting cache access failures which could be attempts to read beyond the size of the file system or cached failures of a decompression failure.

mdmillerii, 2016-04-21:
Todays debug has one system with `free` showing `Active(anon)` growing, free shrinking, and buffered shrinking faster, and another run captured an OOM kill of `/usr/sbin/info` with 
`[232001.260000] Killed process 1027 (info) total-vm:142496kB, anon-rss:139012kB, file-rss:0kB`
On that kill `active_anon:205668kB`.  This `info` command seems to be part of   openbmc/skeleton#59. 

That said, there is probably an issue that squashfs should be able to decompress via a reserved pool in low memory situations for forward progress.  If so that should be a linux issue.  @shenki do you agree?

agangidi53, 2016-04-22:
We are running a test on 6.15 (According to Foxconn, the version with memory leaks fixed). Monitoring the memory pretty close. We can check back on memory profile on Monday to see if it's dying.

agangidi53, 2016-04-25:
We ran the test on 6.15 with Fan Control leak fixes from Ken for 2.5 days. Attached screen shot shows that free memory is still decreasing albeit at a slower rater than previous versions.

[image]

KennethWilke, 2016-04-25:
Looking at our data on this a bit closer, the BMC seems to now lose memory at a rate of about 344KB per hour. The decline in memory is pretty linear, at this trend the BMC would run out of free memory in about 14-15 days.

nkskjames, 2016-04-25:
Would it be worth running base 0.6.1 from github and make sure no memory leaks in base code?

agangidi53, 2016-04-25:
Was thinking the same thing. May be we'll run couple of the servers with base code for 3-4 days to see how the memory looks over there.

KennethWilke, 2016-04-25:
I'll build v0.6.1 and flash it on a box as a base of comparison

KennethWilke, 2016-04-25:
So far on the v0.6.1 branch it's losing about 330KB an hour, though this is only looking at about an hours worth of data

KennethWilke, 2016-04-26:
Here's our chart on the v0.6.1 free memory thus far. The little drop at the end is likely due to my SSHing into the box and running a few things to take a look around.

[image]

KennethWilke, 2016-04-26:
I whipped up a script to monitor the virtual and resident memory usages of all PIDs on the BMC. The script looked at the /proc/{pid}/stat for each process every second and logged increases and decreases. Over the course of an hour I saw a few behaviors:

* My monitor script, hwmon.py and systemd (init) would occasionally allocate 4096 bytes of virtual memory, sometimes a page of resident memory, but would free these allocations resulting in a net increase of 0 bytes over the hour
* systemd-journal increased and decreased it's virtual memory size a few times, but had an overall net increase of 0 bytes of virtual memory
* systemd-journal made several resident memory allocations that were never freed, totaling 548KB over one hour:
  * 107 1-page (4096 byte) allocations
  * 11 2-page (8192 byte) allocations
  * 1 3-page (12288 byte) allocation
  * 1 5-page (20480 byte) allocation

This roughly half-megabyte decrease of memory every hour correlates with the decrease we're seeing in free memory over time. Our box that is running the v0.6.1 base BMC code hasn't run long enough for us to observe what happens when the free memory gets too low, there is a possibility that systemd-journal kicks off some garbage collection routine and that this would be a non-issue. I suppose we'll see in about 7 more days on this particular BMC.

KennethWilke, 2016-04-26:
Also, to add to the above, since busybox's `ps` implementation doesn't show resident memory for each process, this could explain why this issue is easily overlooked.

agangidi53, 2016-04-26:
If this is related to the systemd journal log: Here is the rough output on couple of our boxes. Every 3.5 hours or so it makes a new entry (probably because the buffer fills up ) and then goes to the new one. At any point of time there are about 4 instances of 4MB logs.

 Is it possible that there is some leak in cleaning up and overwriting these logs ? 

root@barreleye:~# ls -l /run/log/journal/0747aeb6c6a64cc3af422d677a30a8e8/
-rw-r-----    1 root     systemd-   4194304 Mar 30 18:32 system.journal
-rw-r-----    1 root     systemd-   4194304 Mar 30 09:23 system@934a2871e5ca48f5b62b78bc093846d9-000000000002089e-00052f3dc399a868.journal
-rw-r-----    1 root     systemd-   4194304 Mar 30 12:53 system@934a2871e5ca48f5b62b78bc093846d9-0000000000021275-00052f40b2a9b979.journal
-rw-r-----    1 root     systemd-   4194304 Mar 30 16:21 system@934a2871e5ca48f5b62b78bc093846d9-0000000000021c4c-00052f43a1adb11e.journal

agangidi53, 2016-04-26:
So looking at above log and chart from Kenneth on 0.6.1:
The first 4 steps in the graph correspond to journal logs (Every 3.5 hours you have a new blob being created):

[image]

KennethWilke, 2016-04-27:
So I let my PID memory watcher script run overnight, looks like every once in a while systemd-journal does dump it's memory once it hits about 4MB or so and rotates the log, which is probably to be expected. We'll keep on eye on it for a bit longer and see if this plateaus at some point, at this point I don't see any specific process that looks to be leaking memory.

KennethWilke, 2016-04-28:
So far the decline persists and remains fairly linear

[image]

causten, 2016-05-03:
KennethWilke, any chance you could share your memory monitor script?

jk-ozlabs, 2016-05-04:
One point: it's fairly normal for the free memory to reduce, as we start filling the buffer cache (free memory is useless to the OS, so the kernel will consume it with cache as appropriate).

Do these charts take this into account? Or is it just the absolute value of "free" memory? As @causten has mentioned, the monitor script source would be helpful too.

agangidi53, 2016-05-04:
Hey @causten and @jk-ozlabs  No the free memory calculator doesn't take into account the buffer cache. We are aware of the fact that this free memory could simply get flushed by a updata / dbflush type process when there isn't enough free memory when another process is in need of it. 

I can share the code with you but it's quite simple, we just parse /proc/meminfo line by line to get the free / Total and Available memory.

For example:
root@barreleye:~# cat /proc/meminfo
MemTotal:         254548 kB
MemFree:           36984 kB
MemAvailable:     133364 kB

agangidi53, 2016-05-04:
So it was buffer cache. Minimum free memory until buffer caching is done was set to 2MB:

_root@barreleye:~# cat /proc/sys/vm/min_free_kbytes
2017
root@barreleye:~# cat /proc/sys/vm/swappiness
60_

So I decided to play around a bit. Checked how much my current free memory is and set the min free memory size to that amount:

_root@barreleye:~# free
             total         used         free       shared      buffers
Mem:        254548       217692        36856        22944        10528
-/+ buffers:             207164        47384
Swap:            0            0            0

**root@barreleye:~# sysctl -w vm.min_free_kbytes=36856
**vm.min_free_kbytes = 36856_

Now i checked the free memory and it exactly dropped to half. (I am not sure why, my expectation was it would stop at the min kb size I set. )

root@barreleye:~# free
             total         used         free       shared      buffers
Mem:        254548       182884        71664        22944         6872
-/+ buffers:             176012        78536
Swap:            0            0            0

agangidi53, 2016-05-04:
This is what setting the min free memory size did to my free memory (Red):

[image]

jk-ozlabs, 2016-05-04:
I suspect that the "roughly double" setting there is the kernel hitting the low watermark (that you've just set), so freeing buffers/cache to the high watermark. From the vm balance docs:

> watemark[WMARK_MIN/WMARK_LOW/WMARK_HIGH]/low_on_memory/zone_wake_kswapd: These
> are per-zone fields, used to determine when a zone needs to be balanced. When
> the number of pages falls below watermark[WMARK_MIN], the hysteric field
> low_on_memory gets set. This stays set till the number of free pages becomes
> watermark[WMARK_HIGH]. 

When you're setting the `min_free_kbytes` sysctl, we set the high watermark to double the low one:

    		zone->watermark[WMARK_LOW]  = min_wmark_pages(zone) + tmp;
    		zone->watermark[WMARK_HIGH] = min_wmark_pages(zone) + tmp * 2;

So, this (particular) observation should just be standard behaviour of the vm system.

Can we get a time series graph of the free, available, buffers, and cache values of `/proc/meminfo` ?

agangidi53, 2016-05-04:
I was just read about watermarks. And was looking to find the watermark calculations to reason the roughly double thing. But you seem to have given that info here anyways.Thanks.

Sure. Let me add that to the monitor, probably tomorrow AM.

jk-ozlabs, 2016-05-04:
So the general strategy here:

- check the resulting data to see if we actually have a memory leak (rather than just filling up caches)
-  if we do:
   - check against the per-process memory figures to see what may be consuming memory; and
   - check the kernel slabinfo to check for a leak in the kernel

KennethWilke, 2016-05-09:
Howdy folks,

I made a couple modifications to the scripts we're running to gather these metrics, to remove hard coded internal endpoints and the like. It's built to push data to graphite specifically though, so you'll need a graphite instance to push data to: https://github.com/KennethWilke/bmc-metric-watcher

To get a graphite up and running quickly, this docker image is pretty nice:
https://github.com/hopsoft/docker-graphite-statsd

jk-ozlabs, 2016-05-12:
Thanks for the scripts @KennethWilke - it looks like they're not taking the caches into account at the moment, which is good to know.

@agangidi53 , are you still able to get a report with the cache & buffer usage too?

And finally, have we actually seen the squashfs bug occur, since fixing the memory leaks discussed above?

KennethWilke, 2016-05-16:
@jk-ozlabs I will add some logic to those scripts to record cache and buffer utilization as well. 

So far we haven't seen any of our BMCs keel over in a couple weeks so at this point I feel safe running this version in the data center and we can probably treat this as fixed, if we see any recurrence we can always reference this issue in a new report.

nkskjames, 2016-05-17:
memory leak is fixed.  BMC has been running for multiple weeks and memory usage is normal

closed 2016-05-17 by nkskjames

Kenthliu, 2016-05-17:
@KennethWilke Just make confirm. Both 0.6.1 stable and 6.15 are fine in the system, right?

jk-ozlabs, 2016-05-17:
@KennethWilke excellent news! Thanks for the update.