openbmc/openbmc #238: modules failing to load at boot, maybe not needed?

causten, 2016-04-24:
Apr 21 17:52:45 barreleye kernel[942]: systemd[1]: Failed to insert module 'ipv6': Function not implemented
Apr 21 17:52:45 barreleye kernel[942]: systemd[1]: Failed to insert module 'kdbus': Function not implemented

Should the yocto build be told to not even try to load these?

shenki, 2016-04-28:
That's systemd being special; these modules do not exist. The error message is harmless.

 - We disable ipv6 support in the kernel. Users of openbmc may wish to enable it, but it saves space in cases we're it's not used
 - Until kdbus gets into the mainline kernel we will not be using it

williamspatrick, 2016-07-27:
1. Need to enable ipv6 in the kernel.
2. Need to make sure systemd is not being built with kdbus.  This is probably already resolved.

msbarth, 2016-08-02:
I checked linux-yocto kernel config and ipv6 is enabled, is there a separate kernel config used for obmc?

amboar, 2016-08-03:
@msbarth yes, the openbmc kernel uses e.g. a configuration at `meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig`. That config currently has `# CONFIG_IPV6 is not set`

msbarth, 2016-08-05:
@causten I checked the system journal on VERSION=0.1.0 and am not able to find those failed module insert entries. It seems that the systemd/kdbus issue has already been resolved, which would explain that entry, however ipv6 is still disabled in the kernel so I would expect to see that journal entry. Could you provide more info on how to check the journal?

causten, 2016-08-05:
boot the system.
journalctl --no-pager > t
grep Failed t
there are a few Failed things.  Point is when there is a problem on a box and I look in the journal to debug I see these messages that add noise

williamspatrick, 2016-08-05:
@msbarth - Keep in mind that qemu and hardware currently use different kernels (in addition to different configs).  You'll need to check on a system to see the issues @causten reported.

msbarth, 2016-08-18:
I did verify this on hardware(barreleye) and did not see those journal entries. Since those journal entries not longer exist and after verifying the AST2400 kernel seems to have IPV6 enabled by default, closing this issue.

closed 2016-08-18 by msbarth