openbmc/openbmc #239: Journald showing errors about lookups missing

causten, 2016-04-24:
Brand new system from MFG.  I captured this trace from dumping the journal.  What are those product numbers and why are they failing

Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'INTERNAL_102') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'INTERNAL_100') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'CHASSIS_102') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'PRODUCT_102') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'MULTI_102') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'MULTI_100') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'INTERNAL_101') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'CHASSIS_101') not found in lookup
Apr 21 17:53:54 barreleye system_manager.py[948]: Caching: /org/openbmc/inventory/system/chassis/hdd_backplane
Apr 21 17:53:54 barreleye system_manager.py[948]: ERROR SystemManager: dbus.String(u'PRODUCT_101') not found in lookup

Level...
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (master)
VERSION_ID=6.12
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (master)
BUILD_ID=v0.6-12-g19a1c40-dirty

anoo1, 2016-04-25:
These are being addressed via:
https://github.com/openbmc/skeleton/pull/60
https://github.com/openbmc/ipmi-fru-parser/pull/14

anoo1, 2016-05-26:
Fixed on above pull requests, please verify in tag v0.8.

closed 2016-05-26 by anoo1