openbmc/openbmc #240: Failed to resolve getPowerState to dbus

causten, 2016-04-24:
I dumped the journald output from a brand new from MFG system. I see this error a lot....  

Apr 21 17:54:08 barreleye system_manager.py[948]: Failed to resolve getPowerState to dbus: The name org.openbmc.control.Chassis was not provided by any .service files

ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (master)
VERSION_ID=6.12
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (master)
BUILD_ID=v0.6-12-g19a1c40-dirty

williamli80, 2016-07-25:
I think it was caused by the mismatch of the code level too. 

After some search, I find out that getPowerState is provided by org.openbmc.control.Power not org.openbmc.control.Chassis.

adamliyi, 2016-07-26:
This error message comes from an old version of Foxconn's branch of skeleton:
https://github.com/SteveLinCH/skeleton/blob/0cc0fe52a96dc4738259f31e62926247c892684a/objects/info.c
The code is not merged to upstream, so I don't think we need to track it.

williamli80, 2016-07-26:
Thx @adamliyi . So I wouldn't find it in our source codes.

vishwabmc, 2016-07-26:
@williamli80 , I was taking to Sivas on a different topic and he pasted some code from the robot automation. I looked into it and saw it had references to getValue. https://github.com/sivassrr/SivasPublic/blob/master/test_sensors_dbus.robot 

I am guessing its from there.

williamli80, 2016-07-26:
Thx @vishwabmc , Those error log came from April 25.  Sivas's test case that you metioned is established on June 14.  So it isn't come from there. But errors may come from other test cases. I'll check.

williamli80, 2016-07-26:
I find getValue is call in openbmc-automation/tests/test_obmcrest.robot#L278  and
openbmc-automation/tests/test_obmcrest.robot#L281.

closed 2016-08-16 by williamli80