openbmc/openbmc #241: Failed to resolve fruid to dbus

causten, 2016-04-24:
I dumped the journald output from a brand new from MFG system. I see this error a lot.... 
Apr 21 17:54:23 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist

Here is the message in context of what else is happening...
Apr 21 17:54:22 barreleye system_manager.py[948]: Running System State: HOST_POWERED_ON
Apr 21 17:54:22 barreleye system_manager.py[948]: Starting process: /usr/sbin/fan_algorithm: fan_algorithm
Apr 21 17:54:23 barreleye system_manager.py[948]: Starting process: /usr/sbin/pcie_slot_present.exe: pcie_present
Apr 21 17:54:23 barreleye system_manager.py[948]: Booting host
Apr 21 17:54:23 barreleye system_manager.py[948]: Checking Presence: SLOT0_RISER_PRESENT
Apr 21 17:54:23 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist
Apr 21 17:54:24 barreleye system_manager.py[948]: GPIO Lookup:  SLOT0_RISER_PRESENT = 424,in
Apr 21 17:54:24 barreleye system_manager.py[948]: Checking Presence: SLOT1_RISER_PRESENT
Apr 21 17:54:24 barreleye system_manager.py[948]: GPIO Lookup:  SLOT1_RISER_PRESENT = 425,in
Apr 21 17:54:24 barreleye system_manager.py[948]: Checking Presence: SLOT2_RISER_PRESENT
Apr 21 17:54:24 barreleye system_manager.py[948]: GPIO Lookup:  SLOT2_RISER_PRESENT = 426,in
Apr 21 17:54:25 barreleye obmc-phosphor-eventd[944]: Low HDD change:0x3,0x0 (assoc)

ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (master)
VERSION_ID=6.12
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (master)
BUILD_ID=v0.6-12-g19a1c40-dirty

williamli80, 2016-07-18:
I think it was caused by the mismatch of the code level.

williamli80, 2016-07-26:
By journalctl --no-page>/tmp/log,, I don't see the error message by now. So I think our upstream code will not bing in that error message. The upstream was refactored.
This is similar to openbmc/openbmc#244.

williamli80, 2016-07-26:
I find getValue is call in openbmc-automation/tests/test_occ_powercap.robot#L119.

closed 2016-08-16 by williamli80