openbmc/openbmc #242: Highest DIMM temperatur is value xx blasting to journald.

causten, 2016-04-24:
Not sure why but I see this text appear a lot (117 times in <2 seconds).  Is this really the best use of journald?  we will lose a lot of information is to continue.  I have no idea what drives this print but wouldn't it make sense to print some if it changes, not stays the same?

Apr 21 18:22:43 barreleye system_manager.py[948]: IMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43
Apr 21 18:22:43 barreleye system_manager.py[948]: Highest DIMM temperatur is value 43

ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (master)
VERSION_ID=6.12
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (master)
BUILD_ID=v0.6-12-g19a1c40-dirty

johnhcwang, 2016-04-25:
That should be the debug message to make sure the highest DIMM temperature for fan speed control. so I helped to mark it on [ef9add](https://github.com/johnhcwang/skeleton/commit/ef9adda0444b7eee7482092454f8179a02fd912e#diff-43a94e52bb1ea291a0a0e57c46aa5d51R419).
We won't see this message in 6.14 and later.

williamspatrick, 2016-06-15:
@bradbishop - Please confirm if this is still an issue.  Commenting out a printf is a trivial commit if needed.

bradbishop, 2016-06-21:
Same thing as https://github.com/openbmc/openbmc/issues/243....this isn't upstream code doing this.

bradbishop, 2016-06-21:
@williamspatrick please close...

closed 2016-06-21 by williamspatrick