openbmc/openbmc #243: PowerState value=[1] blasting in to journald

causten, 2016-04-24:
Not sure why but I see this text appear a lot. Is this really the best use of journald? we will lose a lot of information if this is to continue. I have no idea what drives this print but wouldn't it make sense to print some if it changes, not stays the same?

Apr 21 18:25:55 barreleye system_manager.py[948]: te value=[1]
Apr 21 18:25:55 barreleye system_manager.py[948]: PowerState value=[1]
Apr 21 18:25:55 barreleye system_manager.py[948]: PowerState value=[1]
Apr 21 18:25:55 barreleye system_manager.py[948]: PowerState value=[1]
Apr 21 18:25:55 barreleye system_manager.py[948]: PowerState value=[1]

ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (master)
VERSION_ID=6.12
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (master)
BUILD_ID=v0.6-12-g19a1c40-dirty

johnhcwang, 2016-04-25:
That should be the debug message to make sure the current power state in info. so I helped to mark it on [ef9add](https://github.com/johnhcwang/skeleton/commit/ef9adda0444b7eee7482092454f8179a02fd912e#diff-6c21597f2199eea0c40b2028732d1923R244).

williamspatrick, 2016-06-15:
@bradbishop - Another simple cleanup if still there.

bradbishop, 2016-06-21:
I can't find this in the code anywhere and I don't see it with newer builds.  Closing...

bradbishop, 2016-06-21:
I can't close.  And I understand why I can't see it in the code - this is in code that was on the Barreleye when it shipped but it hasn't been upstreamed.

closed 2016-06-21 by williamspatrick