openbmc/openbmc #244: ipmi failing on dbus communication

causten, 2016-04-24:
I added a few lines before and after but it seems @56:23 there was a call to dbus without an interface/method/path.  

Apr 21 17:56:14 barreleye system_manager.py[948]: Running System State: HOST_BOOTED
Apr 21 17:56:15 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist
Apr 21 17:56:17 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist
Apr 21 17:56:20 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist
Apr 21 17:56:22 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist
Apr 21 17:56:23 barreleye ipmid[961]: WATCHDOG ERROR:[0x80] handling NetFn:[0x2C], Cmd:[0x3]
Apr 21 17:56:23 barreleye system_manager.py[948]: ERROR SystemManager: 59 not found in lookup
Apr 21 17:56:23 barreleye system_manager.py[948]: ERROR SystemManager: 59 not found in lookup
Apr 21 17:56:23 barreleye ipmid[961]: Failed to create a method call: Invalid argumentBus:  Path:  Interface: org.openbmc.InventoryItem
Apr 21 17:56:23 barreleye ipmid[961]: Failed to get a fru type: Invalid argumentBus: , Path: , Interface: org.openbmc.InventoryItem
Apr 21 17:56:23 barreleye ipmid[961]: ERROR:[0xCB] handling NetFn:[0x4], Cmd:[0x2D]
Apr 21 17:56:24 barreleye system_manager.py[948]: Failed to resolve fruid to dbus: Method "getValue" with signature "" on interface "org.openbmc.SensorValue" doesn't exist
Apr 21 17:56:24 barreleye system_manager.py[948]: Installing OCC device
Apr 21 17:56:24 barreleye system_manager.py[948]: Installing OCC device

williamli80, 2016-07-04:
I find it in ipmi-fru-parser repository's commit
[https://github.com/openbmc/ipmi-fru-parser/tree/f3ca35258caec338c72972559ca6b20a2df920f1](url)

commit f3ca35258caec338c72972559ca6b20a2df920f1
Author: vishwa <vishwanath@in.ibm.com>
Date:   Wed Dec 2 10:35:13 2015 -0600

    set present status True after updating inventory

williamspatrick, 2016-07-05:
@williamli80 - That URL doesn't work for me.  

You can reference a commit like this
```
openbmc/ipmi-fru-parser@f3ca35258
```
openbmc/ipmi-fru-parser@f3ca35258

williamspatrick, 2016-07-05:
@williamli80 - In the current code, and even the commit you referenced, the method call was getObjectFromId and not getValue.  I don't think this is where the message is coming from.
    https://github.com/openbmc/ipmi-fru-parser/blob/master/writefrudata.C#L189

williamli80, 2016-07-15:
By journalctl --no-page>/tmp/log,, I don't see the error message.

williamli80, 2016-07-25:
After searching those repositories, I cannot find where the getObjectFromId was called.

vishwabmc, 2016-07-25:
https://github.com/openbmc/ipmi-fru-parser/blob/master/writefrudata.C#L189 is where the getObjectFromId is called @williamli80  but that is not what was reported as the problem tho ?.

The call is defined in : https://github.com/openbmc/pyphosphor/blob/a88b3239b0a9592defaeee59c0ff17253a86aa39/obmc/sensors.py and the guy that is interested in must be the IPMI sensor handlers but I don't see it getting called.

To me, it looks like it was part of some code that is already re-factored. ??

williamli80, 2016-07-26:
@vishwabmc  Thanks,  I made a typo. I mean I cannot find where getvalue is called.

closed 2016-08-16 by williamli80