openbmc/openbmc #249: BMC build tar ball generation: Missing files for some partitions

agangidi53, 2016-04-28:
Hi All

2 things:
**First:
****According the BMC users guide 0.6: 
 [system_name]-[datetime].tar: contains (6) individual images for:
1. u-boot, 
2. uboot settings (u-boot-env), 
3. initramfs, 
4. kernel 
5.  R/W file system overlay (rwfs),
6.  the read-only filesystem (rofs). 

But According to the latest build I generated it contains only only 5 images (u-boot-env partiton is not present):
1. image-rwfs 
2. image-initramfs
3. image-kernel
4. image-rofs
5. image-u-boot

**Second**:
**There are 8 mtd partitions in total in BMC booted up, Why are only 5 images generated in the tar ball ? : 
root@barreleye:/dev# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 02000000 00001000 "bmc"
mtd1: 00060000 00001000 "u-boot"
mtd2: 00020000 00001000 "u-boot-env"
mtd3: 00280000 00001000 "kernel"
mtd4: 001c0000 00001000 "initramfs"
mtd5: 01740000 00001000 "rofs"
mtd6: 00400000 00001000 "rwfs"
mtd7: 04000000 00001000 "pnor"

I understand the "all" image might have content of all partitions, but if someone wants to use the individual images to manually flash, won't they miss the changes in rest of partitions if they use the regular tar ball with individidual images ?

nkskjames, 2016-04-29:
bmc is the full image. pnor is the pnor which we don't build. We currently don't build a set of default u-boot-env.   When preserving the uboot environment variables, before the flash operation, we copy the current values to the u-boot-env partition to preserve them.

williamspatrick, 2016-06-17:
@agangidi53 - Can you give an update on where we are at with this one?

agangidi, 2016-07-01:
Hey @williamspatrick   

I want to be able to generate a uboot-env partition and flash it along with all the partitions instead of env getting generated on first boot.

gtmills, 2018-05-23:
@agangidi53 @anoo1 Is this still an issue?

agangidi53, 2018-05-31:
Although what I was looking for is still not implemented but feature needs goes down as we've worked around the "mac address sync" issues quite nicely. 

This can be closed.

closed 2018-05-31 by gtmills