openbmc/openbmc #257: RestAPI curl response 443: connection refuse

Kenthliu, 2016-05-06:
The curl command get 443 response. This issue happen after stress curl command or DC cycle test.
The BMC can be still ping. SSH is OK. Port:3000.
[image]
Attach the top data and journalctl log.
[image]

[log.txt](https://github.com/openbmc/openbmc/files/252240/log.txt)

nkskjames, 2016-05-10:
@Kenthliu there should be a process called 'obmc-rest' running.  Can you do a 'ps | grep rest' and see what is running?

Kenthliu, 2016-05-10:
[curlfail.txt](https://github.com/openbmc/openbmc/files/256326/curlfail.txt)

nkskjames, 2016-05-10:
The failing curl commands are not in that log.   I was hoping to also see the error message when you tried to restart obmc-rest process.

bradbishop, 2016-05-10:
Apr 15 09:26:46 barreleye systemd[1]: obmc-mapper.service: Start operation timed out. Terminating.
Apr 15 09:26:46 barreleye systemd[1]: Failed to start Phosphor OpenBMC DBus service management daemon.
Apr 15 09:26:46 barreleye systemd[1]: Dependency failed for Phosphor OpenBMC DBus REST daemon.
Apr 15 09:26:46 barreleye systemd[1]: obmc-rest.service: Job obmc-rest.service/start failed with result 'dependency'.
Apr 15 09:26:46 barreleye systemd[1]: obmc-mapper.service: Unit entered failed state.
Apr 15 09:26:46 barreleye systemd[1]: obmc-mapper.service: Failed with result 'timeout'.

bradbishop, 2016-05-10:
I don't even want to think about how long it would take to introspect the 1200 event log objects....

I have a change coming that should help here - rather than doing 1200 introspections it will just be one dbus call to GetManagedObjects.

Even still....1200 event logs seems excessive.  Would it make sense to have a cap?

nkskjames, 2016-05-10:
probably makes sense to have a cap.   so sounds like we have 3 issues:
1. there is something wrong with the hardware that this test was ran on.  it is generating a ton of event logs
2. need fix for using GetManagedObjects
3. need a cap on event logs (or lower size allocation).

KennethWilke, 2016-05-10:
Based on the 'connection refused' message, it looks like the service that was listening on port 443 may have died, when this happens again the results of `netstat -tnl` should show what ports are still being listened on.

Kenthliu, 2016-05-11:
@bradbishop So do you think the root cause is related to the event over flow?

anoo1, 2016-05-17:
1. there is something wrong with the hardware that this test was ran on. it is generating a ton of event logs - Ken, is this something your team can take a look from the event logs, or perhaps provide some of the logs?
2. need fix for using GetManagedObjects - https://github.com/openbmc/phosphor-objmgr/pull/11
3. need a cap on event logs (or lower size allocation). - https://github.com/openbmc/phosphor-event/pull/18

johnhcwang, 2016-05-18:
Regarding to the question 1, we expect that is a normal event which comes from HDD info service.
Every AC cycle it will add this event to tell user how many HDD was detected.

bradbishop, 2016-05-26:
I think this should be fixed by https://github.com/openbmc/phosphor-objmgr/pull/11

@Kenthliu can you try this again?

Kenthliu, 2016-05-26:
OK we will verify in 0.8.1. Thanks.

Kenthliu, 2016-05-26:
@bradbishop I think openbmc/phosphor-objmgr#11 should include in 0.8.1. I make the event log full, but the connection still show 443. I did not see obmc-rest and phosphor-mapper show in ps. After I clear the log and AC cycle, the restapi connection work. Attach the journctl log.
[LANerror.txt](https://github.com/openbmc/openbmc/files/283647/LANerror.txt)

causten, 2016-06-07:
Assign to me too please

causten, 2016-06-10:
I'm trying on 0.8.1 tag to recreate.  So far no luck.  I can access the REST interface each time.

### This is how I created the 128 test logs...
#!/bin/bash 
COUNTER=0
busctl call org.openbmc.records.events /org/openbmc/records/events org.openbmc.recordlog clear   
while [  $COUNTER -lt 128 ]; do
    busctl call org.openbmc.records.events /org/openbmc/records/events org.openbmc.recordlog acceptTestMessage
    let COUNTER=COUNTER+1
done

#  I was able to access the BMC with this command....
curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://x.x.x.x/login

Kenthliu, 2016-06-13:
@causten Sorry the reproduce step not so clear. I use the web dbus port:3000 to add the test logs. Can you please try again? Thanks.

anoo1, 2016-06-20:
A fix for number of event logs was done with: https://github.com/openbmc/phosphor-event/pull/19
In addition, Milton believes that this could be fixed with the u-boot memory fix available now in openbmc tag 1.0. We should check that tag 1.0 fixes this issue.

Kenthliu, 2016-06-24:
I verified in 1.0. I think it's OK. This issue can be closed.

closed 2016-06-24 by Kenthliu

sivassrr, 2016-06-27:
sivas@ubuntu-siva:~$ ssh root@9.126.171.197
root@9.126.171.197's password:
root@palmetto:~# cat /etc/version
20160627011830
root@palmetto:~# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0-13-gd6abfb9-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0
root@palmetto:~#

Out of 212 TCs, only 10 passess, Due to Errno 111 - connection refused (http connection)

sivassrr, 2016-06-27:
http://9.126.171.197:3000/ - works well on the failure system.

Kenthliu, 2016-06-28:
1.0.1 is also looks good now. We can close this issue.