openbmc/openbmc #258: Security Feature: Lockdown BMC chip

anoo1, 2016-05-09:
This is a customer request agreed by to lockdown the BMC chip from users for security reasons.

anoo1, 2016-05-09:
Per previous discussions, this may need input from the IBM's LTC team to see if an interim implementation could be done sooner while the full lockdown feature is designed.

williamspatrick, 2016-06-15:
This is not necessarily applicable to Witherspoon.  Marking as low priority for now.

geissonator, 2019-02-25:
@amboar or @anoo1 any idea on what this one was tracking? Given it's age and our recent changes in this area, I assume we can just close this one out?

amboar, 2019-02-25:
the "from users" is the unclear part. @anoo1 - what did that mean?

amboar, 2019-03-19:
No-one cares, and given recent developments this has probably been taken care of.

commit 4905e8cf (2021-07-27): meta-google: sslh: Bump 517e4ad..de0ec95

yrutschle (25):
      changelog the sslh-select refactoring
      justify use of select
      remove some debug messages
      made echosrv independant from common.o and with its own configuration
      add UDP support to echosrv
      when verbose, precise which listen sockets are udp
      remove caveat about UDP refactor
      upgraded argtable3 to current last version to fix a parsing issue with command line options
      do not drop CAP_NET_ADMIN if any of the protocols require transparent proxying
      typo
      refactor: purified set_capabilit [... cut]
closed 2019-03-19 by amboar