openbmc/openbmc #260: Implement PAM Authentication

anoo1, 2016-05-09:
Some initial work found that adding the PAM package to openbmc makes the image too big to fit on flash. Also it turns off busybox. So this work would need some recipe manipulation to only add the desired PAM package to the image and prevent busybox from turning off busybox.

amboar, 2016-05-10:
@anoo1 can we add some words explaining the motivation for introducing PAM? How do we plan to use it?

anoo1, 2016-05-10:
@bradbishop will be able to provide a better background for implementing PAM.

williamspatrick, 2016-06-15:
Should include:

1. Updating PAM recipe to ensure it fits on flash.
2. Converting login applications to use PAM.

williamspatrick, 2016-09-07:
Created #545 for a portion of this.

zahrens, 2017-11-16:
PAM based authentication (Total: 2pts)
a) Deploy PAM config files.
b) What does systemd have to offer in terms of PAM? Do we need to implement our own plugin?
d) Login implementation to use PAM. (2pts)

rfrandse, 2018-08-17:
moved to https://github.com/ibm-openbmc/dev/issues/86

closed 2018-08-17 by rfrandse