openbmc/openbmc #261: Implement Remote Media Support

anoo1, 2016-05-09:
This is a work in progress.

amboar, 2016-05-10:
@anoo1 can you provide more detail here? What media are we exposing? What's the motivation behind the work? Also, it sounds like someone's taken up the task and I think it would be useful to set the Assignee field so we know who's doing what.

anoo1, 2016-05-10:
@mdmillerii  has being working on this item, he would be able to provide more details.
Don't have permissions to set the Assignee, will ask Patrick to assign.

mdmillerii, 2016-06-30:
The plan for this item is to support remote media by presenting a virtual CD drive over the USB2.0 vhub as a usb gadget device.

mdmillerii, 2016-06-30:
Current status is I have tried a minimal forward port of the code in facebook/openbmc but writes to the reset register in the vhub were not taking effect.  I had the minimum 10us delay for clock start but did not verify strapping registers or pin muxes for usb clock and 24 or 48MHz refclock.

closed 2018-08-16 by rfrandse

anoo1, 2018-08-16:
This has been moved to https://github.com/ibm-openbmc/dev/issues/63