openbmc/openbmc #262: Refactor openbmc/skeleton

anoo1, 2016-05-09:
The openbmc/skeleton repository is a temporary one. Need to re-implement its interfaces in the permanent locations.

shenki, 2016-05-10:
This needs to include

 - remove all use of `/dev/mem` in favour of kernel interfaces
 - remove the copy of `pflash` from skeleton
 - instead use https://github.com/open-power/skiboot/tree/master/libflash for flash operations, taking advantage of the mtd support

williamspatrick, 2016-09-12:
Pflash has been removed and libflash is used instead.

openbmc/skeleton@26be06904f383fd0a323713e0c3c9dbaec146b81

geissonator, 2016-12-01:
Some notes from my analysis of a lot of the skeleton objects:

bmcctl: /org/openbmc/control , org.openbmc.control.Bmc , bmc0
- Registers for 2 signals, handle-init and handle-warm-reset
-- handle-init just calls control_complete_init (goes into the big interface file)
-- handle-warm-reset is where we do the “sleep 3; reboot” call.

hostcheckstop: /org/openbmc/control , org.openbmc.control.Checkstop , checkstop0
- Monitors for checkstop GPIO and initiates chassis_reboot and verifies host booted
-- Uses getSystemState in pysystemmgr, looking for “HOST_BOOTED” or “HOST_BOOTING"

hostwatchdog: /org/openbmc/watchdog , org.openbmc.watchdog.Host , host0
- Handles all watchdog signals
-- handle-start
-- handle-poke
-- handle-stop
-- handle-set

op-hostctl: /org/openbmc/control , org.openbmc.control.Host , host0
- This is the code that does the FSI bit banging, it looks for 2 signals.  It also defaults debug_mode to 0, and the host flash side to “primary"
-- handle-boot: Check debug_mode, only enable fsi and cronus otherwise bit bang the host up and sed/start the watchdog
-- handle-init: just calls control_complete_init (goes into the big interface file)

op-pwrctl: /org/openbmc/control , org.openbmc.control.Power , power0
- Monitors for certain signals and sets power gpio pins accordingly.  Responsible for power on/off and pgood monitoring
- It starts pgood monitoring loop when it first starts, sets system state accordingly (goto-system-state HOST_POWERED_OFF/ON), and does the pcie and usb gpio resets as needed
- This code has a sub-dir, pgood_wait which has a standalone app to monitor for pgood
-- handle-set-power-state: Writes the gpio to power on or off
-- handle-get-power-state: Return pgood state
-- handle-init: just calls control_complete_init (goes into the big interface file)

pciedetect: Determines PCIe device presence and updates corresponding inventory objects

pwrbutton: /org/openbmc/buttons , org.openbmc.buttons.Power, power0
- Subscribes to GPIO interrupt, emits signals for button pressed/released
- Monitors for signals
-- handle-is-on:
-- handle-sim-press: 

pychassisctl: org.openbmc.Control , org.openbmc.control.Chassis , chassis0
- Monitors for multiple signals
-- power button released, pressed long, soft reset
-- watchdog error
-- emergency shutdown
-- system state changes (GotoSystemState) - if HOST_POWERED_OFF or ON, set system_state in org.openbmc.settings.Host
--- if it’s a HOST_POWERED_OFF and the internal “reboot” flag is set to 1 then power on
- Provides multiple interfaces
-- LED identify and clear
-- power on and off (systemd calls - objc-chassis-start@0.target and stop) (sets internal “reboot” flag to 0)
-- softPowerOff (host services pass through)
-- reboot (if power state is off, power on - else power off)
-- softReboot (if power off, power on - else do soft power off)
-- getPowerState (pass through to getPowerState from pwrctl)

pystatemgr: 
- Looks at pgood from org.openbmc.control.Power and sets BootProgress sensor and occstatus0/1 sensors
-- otherwise if pgood is off, it gets the last state from org.openbmc.managers.System, looks at the power_policy from or.openbmc.settings.Host and if needed, it calls the powerOn interface in pychassisctl.

pysystemmgr: org.openbmc.managers.System , /org/openbmc/managers/System
- Maintains the system state, by monitoring watching new interfaces showing up on dbus and by providing a GotoSystemState
- Keeps track of 2 parameters
-- current_state (i.e. HOST_BOOTING) - method getSystemState()
-- system_last_state (i.e. 0 (powered off, 1 (powered on)) - it caches this to filesystem
- This code uses the SYSTEM_STATES from the config files (i.e. openbmc/skeleton/configs/Barreleye.py) to transition states
-- Either someone signals via GoToSystemState or specified objects in EXIT_STATE_DEPEND have started
-- Or it sees the org.freedesktop.Dbus.Properties change signal for org/openbmc/control/power0

pydownloadmgr: org.openbmc.managers.Download , /org/openbmc/managers/Download
- Looks like the tftp support

pyfanctl: org.openbmc.control.Fans , /org/openbmc/control/fans
- Kind of looks like it just max’s out fans?

pyflashbmc: org.openbmc.control.BmcFlash , /org/openbmc/control/flash/bmc
- Handles updating the BMC flash

pyhwmon: org.openbmc.Sensors , /org/openbmc/sensors
- hwmon function

mine260309, 2016-12-15:
What about the refactor of GPIO related configs?
Note that a couple of skeleton code uses the GPIO related code in `libopenbmc_intf`:
  * hostcheckstop
  * rstbutton
  * pciedetect
  * op-pwrctl
  * op-hostctl
  * pwrbutton

When refactoring, it will have dependency to it, so it seems to me that GPIO refactor shall be done before such projects?

geissonator, 2017-05-23:
Remaining org.openbmc items:
[witherspoon]# busctl tree | cat | grep org.openbmc | grep Service
Service org.openbmc.NetworkManager:
Service org.openbmc.control.Power:
Service org.openbmc.managers.System:
Service org.openbmc.control.Checkstop:
Service org.openbmc.records.events:
Service org.openbmc.Sensors:
Service org.openbmc.buttons.reset:
Service org.openbmc.control.Fans:
Service org.openbmc.mboxd:
Service org.openbmc.managers.Download:
Service org.openbmc.settings.Host:
Service org.openbmc.control.Chassis:
Service org.openbmc.buttons.Power:
Service org.openbmc.Inventory:
Service org.openbmc.control.BmcFlash:
Service org.openbmc.control.Flash:
Service org.openbmc.HostIpmi:
Service org.openbmc.TimeManager:
Service org.openbmc.UserManager:
Service org.openbmc.control.Bmc:
Service org.openbmc.watchdog.Host:

anoo1, 2017-05-24:
> Service org.openbmc.records.events:
> Ready to be removed

I committed to removing this, will do in the next few days.

> Service org.openbmc.mboxd:
> Handled in #1439 and related changes or remove story needed?

@geissonator Please open a story to rename the service file, this is not an 'old' mbox but the new and only one :)

> Service org.openbmc.control.BmcFlash:
> #349 has eventual removal story or one needed?
> Service org.openbmc.control.Flash:
> #349 has eventual removal story or one needed?

Covered with #1614 

> Service org.openbmc.managers.Download:
> #349 has eventual removal story or one needed?

I added a comment to remove with #1614

geissonator, 2017-05-24:
All known sub-stories created, moving this epic out to phase 6 and reducing to 5pts

geissonator, 2017-08-02:
Leave 1 pt for anything we missed

closed 2019-10-27 by stale[bot]