openbmc/openbmc #265: Openbmc Redfish Compliant

anoo1, 2016-05-09:
Track the work needed to make openbmc Redfish compliant

vrpradeepkumar, 2017-08-01:
Hi,

Is there any plans regarding this?
Interested  to contribute in Redfish support.
If there is any plans/design or implementation which is already done regarding this please let me know.

Currently planning to support the network URI (redfish/v1/Managers/bmc/EthernetInterfaces/eth0) in Redfish.

Thanks
Pradeep

mdmillerii, 2017-09-02:
There is some outline documentation talking about the requirements of an implementation in gerrit from @causten.

https://gerrit.openbmc-project.xyz/#/c/5792/

krtaylor, 2019-03-18:
Firmware update design:
https://gerrit.openbmc-project.xyz/#/c/openbmc/docs/+/18186/

gtmills, 2019-11-16:
The following Redfish functionality was in 2.7: 
Firmware Update
Inventory
Logging
Network
Partial Certificates
Power & Thermal
State Management
Time
Management
User Management

The following Redfish functionality is planned for 2.8 (subject to change):

All Sensors
Event Service
Full Certificate Management
LDAP
LED
Multiple Firmware Image Support 
Telemetry Service
Subscription Collection
Virtual Media

gtmills, 2019-11-16:
Closing this issue since 2.7 (warrior) was released in Aug 2019.

closed 2019-11-16 by gtmills